MJoergen/C64MEGA65 #124: Lazarian Cartridge does not work on R5 but works on R3

sy2002, 2024-01-26:
Reported by @dansanderson here https://github.com/MEGA65/mega65-core/issues/778#issuecomment-1909362620:

Tested with Alpha 5:

R5: Lazarian: screen goes black, or stuck on blue background light blue border
Works on R3.

Question for @dansanderson do you have a modded or unmodded R5?

dansanderson, 2024-01-26:
Currently un-modded. Once I get a KCS and repro the breakage I'll do the 7-wire mod and try again.

sy2002, 2024-01-26:
Thank you @dansanderson - just for clarity: We already tested KCS on modded R5s: Works like a charme. We are more interested in you testing Lazarian with a modded R5.

dansanderson, 2024-01-26:
Understood, I just mean I want evidence that the mod is doing what it's supposed to, so repro'ing a known broken case that it fixes would be helpful. :)

dansanderson, 2024-01-27:
Lazarian just started up for me on R5, played fine. I think we can close this one.

sy2002, 2024-01-27:
Thank you Dan, closing the issue.

closed 2024-01-27 by sy2002