MJoergen/C64MEGA65 #125: Q-Bert Cartridge does not work on R3 and R5

sy2002, 2024-01-26:
Reported by @dansanderson here https://github.com/MEGA65/mega65-core/issues/778#issuecomment-1909362620

"Q-Bert: player select works, then screen goes black

Same behavior on R3, might be a bad cart."

dansanderson, 2024-01-27:
I got this cart working on my Ultimate 64, so there might be a genuine issue. I will re-test on the R5 C64 core before and after the wire mod.

dansanderson, 2024-01-27:
Can still repro. Also for what it's worth the MEGA65 core on R5 with GO64 mode gets further: displays the board and plays the opening them, then hangs. 😅 We're not here to troubleshoot GO64 mode but this might be relevant with regards to R5 cartridge port wiring.

sy2002, 2024-01-27:
Thank you @dansanderson . We did not find Q-Bert on eBay. We might either come up with some remote-debugging setup, where you might want to install Vivado and follow our instructions deliver "ILA measurements" - or we might need to find a way to get our hands at a Q-Bert cart.

Just for my understanding - because this is important to decide if this would block the 5.1 release of the core:

Is this very cartridge working with the V5 version of the C64 core on an R3 board and not working on the V5.1 version on an R5 board or is not not working on both boards and core revisions? (The initial message - scroll up - suggests the latter.)

In the latter case, if it is not working neither on R3 using **V5** of the core nor on R5 using **V5.1**, then we would leave this issue open and tackle it at a future release (V6 or so) and maybe we'd be able to buy a Q-Bert cartridge until then.

But if it DOES work on V5 / R3 but not on V5.1 / R5, then we might consider this as a "red alert" sign, because this might be just the tip of an iceberg.

dansanderson, 2024-01-27:
Confirmed: Q-Bert has the same issue on C64 core V5 on R3, as well as C64 core V5.1 on R3 and V5.1 on R5. This is not a regression.

We should also take into account that this is probably a PAL/NTSC thing. This cart runs on NTSC machines so might not work at all for PAL, in which case it's not an issue at all. I don't have a vintage PAL C64 to test with, only the C64 core (PAL) and vintage NTSC C64s.

sy2002, 2024-01-29:
Thank you @dansanderson: Indeed it is very likely that this is a NTSC vs PAL issue. I will leave this issue open, because this very Q-Bert cartridge will be a nice test for us, as soon as we will have added NTSC support to the C64 core and I want to avoid that this info falls through the cracks. NTSC support is on our roadmap, but not yet for the up and coming 5.1 release. So I changed the "bug" label to "research" and added the "V6 or later" label.

cwscws, 2024-04-05:
I tried Q-Bert as PRG and it shows the same behavior. Select screen works then goes black. Same file works perfectly on any C64 emulator.

sy2002, 2024-04-05:
Thank you @cwscws : Might be a PAL vs NTSC issue and needs more investigation. Does not block the V5.1 release as this issue was also present in V5.

cwscws, 2024-04-06:
Only using PAL. Works also perfect on MiSTer C64 core.

sy2002, 2024-04-07:
@cwscws Thank you! This is interesting feedback! Did I understand you correctly: You switch MiSTer to PAL and then you load Q-Bert as PRG and this works, while the same on our core does not? I assume you are talking about the newest MiSTer?

@paich64 Can you please confirm this on a V4 core and a V5 and a V5.1 RC2 core? On a R3A and on your prototype R5?

cwscws, 2024-04-08:
I’m using the C64 core on MiSTer (latest 2024 version) in PAL mode only. Loading the PRG via the menu like on Mega65. Just works and can play. On the C64 core for Mega65 after selecting 1 or two players the scrren goes black and nothing happens anymore. Tried with regular kernel and with JiffyDOS but this does not affect anything.

MJoergen, 2026-04-21:
Fixed in Alpha Release V6.0A6