MJoergen/C64MEGA65 #126: Support Kernals in EF3

sy2002, 2024-01-26:
To support Kernals in the Easy Flash 3 (EF3), we need to be able to sense the cartridge's modifications of address line A14 as described here:

"Fortunately, a method was found which does not involve #DMA. Whenever a CPU read
access to the address range $F000-$FFFF is detected, the cartridge pulls down the address
line A14 for a fraction of the clock cycle. This changes any address between $F000-$FFFF
to an address between $A000-$BFFF. After this is done, the state of #HIRAM can be read
from #ROMH."

Taken from: http://skoe.de/kernal/kernal-cartridge.pdf

There are two ways of doing this that I currently see:

1) Enhancing our heuristics module (https://github.com/MJoergen/C64MEGA65/blob/develop/CORE/vhdl/cartridge_heuristics.vhd) and this code here: https://github.com/MJoergen/C64MEGA65/blob/develop/CORE/vhdl/main.vhd#L991  (This means: We monitor in realtime what the firmware code of the EF3 is doing and then act accordingly. We are already doing this today on R3/R3A boards, otherwise the EF3 would not work at all.)

2) Enhancing future MEGA65 PCBs to have circuit that allows us to continue to constantly write to the address bus while in parallel we can sense if the EF3 modifies A14

Advantage of (1): No hardware change needed plus works on all MEGA65 revisions. But I consider this as pretty dangerous, see comments in main.vhd (see link above).

Advantage of (2): Clean solution

gardners, 2024-01-26:
The third way:
At the right time during each phi2 cycle when the KERNAL cartridge pulls A14 low to probe HIMEM, switch the A8-15 bank to input on the MEGA65 expansion port, and read A14: If it has gone low, then respond as a C64 would in this case, so that the KERNAL cart performs a KERNAL read, and then use that result.

sy2002, 2026-09-13:
Not in scope for V6

sy2002, 2026-11-05:
@kugelblitz360 You wrote about Kernal replacements in #259 - so I wanted to share with you: This issue tracks the effort. I do know pretty much exactly what needs to be done to support it, I only shied away in past due to not having coding agents. Do you consider this feature as important for V6? I could try - but it increases the attack surface for bugs...

Kugelblitz360, 2026-11-06:
Well, the request to support more Kernal ROMs is valid. The safe route would be to extend the current Jiffy feature to load any other Kernal. This is MUCH safer than doing this via the bus. On the other hand, for bragging rights, this completes the cart port recreation.

I'd love to be able to have SOME way to use different Kernals and EF3 as well TR now have this feature.

Your call - if it you think it makes the cart bus unstable, don't do for V6, your time is limited. If you could extend Jiffy to more ROMs, awesome.

Kugelblitz360, 2026-11-06:
https://github.com/MJoergen/C64MEGA65/issues/152 - contains just loading Kernals from the SD Card instead of using external replacements.