MJoergen/C64MEGA65 #128: Zeta Wings cartridge does not work on R5 but works on R3

sy2002, 2024-01-26:
@paich64: You reported this via Skype. Before we can investigate: Can you please confirm:

✅  Zeta Wings works on Version 5 of the core on your R3 board
✅  Zeta Wings works on Version 5.1 RC1 on your R3 board (or with any Alpha version, specify which)
❌  Zeta Wings does not work on Version 5.1 on your R5 board

We don't have full clarity what to test. As soon as you provided details, we will investigate.

paich64, 2024-01-27:
@sy2002 @MJoergen after having investigated a long time yesterday, The conclusion is that it works on both R3 & R5 as i formerly reported. The problèm is that my cartridge in indeed faulty. I tested it yesterday on a C64C and it failed the exact same way as on R3/R5 : I guess you know what it means when a Cartridge is "slowly dying" : sometimes it works, sometimes it doesn't which is very misleading (especially when it's a modern cartridge you would not expect to fail). A friend of mine has investigated the cartridge, unsoldered the ROM support, cleaned up all soldering points and made fresh soldering. He just confirmed to me that the cartridge is back to life and is now working again on the C64C real hardware. 

As i said it had already worked on R3 & R5, but because it started to fail randomely, i though the core was not handling is correctly anymore. Testing it on a true C64 gave us the root cause : Faulty cartridge.

sy2002, 2024-01-27:
Thank you @paich64 - These are great news. I will close the issue.

closed 2024-01-27 by sy2002