MJoergen/C64MEGA65 #129: Feature request: add MONITOR function to freezer menu

MauriceMEGA65, 2024-02-03:
If anyhow possible, I would love to have a built-in ML Monitor. Somewhere in the Freezer menu or under a key combination.
Preferably a Monitor like in the Super Snapshot because it display symbols for a wider range of hex values than SuperMon64.

sy2002, 2024-02-04:
@MauriceMEGA65 Thank you for the feature request. Indeed I was also thinking about something like this. Please open this file and scroll down to the very last line: https://github.com/MJoergen/C64MEGA65/blob/master/ROADMAP.md  . Quoting:
`Hardware debugger (single-step the CPU via the on-screen-menu)`

So yes, this is a really cool idea and we will consider it when time is ripe.

sy2002, 2026-10-09:
Just a small side project: Can we vibe-code a super cool monitor with disassembler and stuff and freeze capabilities (as long as one points us to a d64 or d81 disk image with enough empty space) and then you can mount also the disk image and load the frozen stuff?

sy2002, 2026-11-11:
Out of scope for V6