MJoergen/C64MEGA65 #133: C64 Emulator Test Suite V2.15 fails on TRAP16 test and HANGS on TRAP17 test with V5.1RC2 / V5.1RC1

paich64, 2024-03-13:
C64 core regression tests include running the C64 Emulator Test Suite V2.15.

- V5.0 test suite results reports TRAP1 to TRAP16(included) are OK 
- V5.0 test suite results reports TRAP17 fails, however we can move forward with next tests 

HOWEVER:

- V5.1RC2/V5.1RC1 test suite results report TRAP16 is failing  
- V5.1RC2/V5.1RC1 test suite results report TRAP17  fails but also hangs when displaying debug info. we can't move forward with next tests 

**TRAP16 test issue**
- On **V5.0** **TRAP16** is successfull:  
[image]

- On **V5.1RC2/V5.1RC1** **TRAP16** is failing:
[image]

**TRAP17 test issue**
- On **V5.0** **TRAP17** fails, however we can move forward with next tests :
[image]
[image]

- On **V5.1RC2/V5.1RC1** **TRAP17** fails but also hangs when displaying debug info. we can't move forward with next tests:
[image]
[image]

additional observations for TRAP16 and TRAP17 tests :

- both issues do not occure with latest Mister C64 core 

- The unexpected outcome is exactly the same on R3A and R5 

- Using a lower resolution when testing does not change the outcome

- Disabling flicker free does not change the outcome

To reproduce the issue :

- You don't need to run the entire test suite 
- Mount Test Suite disk 2
- LOAD"TRAP",8,1
- RUN

C64 Emulator Test Suite V2.15 download : 
[tsuit215.zip](https://github.com/MJoergen/C64MEGA65/files/14594512/tsuit215.zip)

C64 Emulator Test Suite V2.15 source code for individual tests:
[testsuite-2.15.tar.gz](https://github.com/MJoergen/C64MEGA65/files/14594516/testsuite-2.15.tar.gz)

sy2002, 2024-03-14:
@paich64 Thank you VERY MUCH for this thorough analysis, the source code, the tests itself, etc.

@MJoergen and I are meeting on Saturday. We will assess, if we see a path ("semi-low hanging fruit") to fix this for the 5.1 release or not. This is why I did for now stick a "V6 or later" not this issue, even though it is clearly a regression.

@paich64 Your demo deep dive testing and your cartridge (.crt and hardware) testing is now twice as important: Because if any additional regressions are now introduced - like a demo not working any more or a cartridge (sw/hw) - then we might want to boost the priority of this one to "5.1 (R4/R5 surge)".

sy2002, 2024-03-14:
From "Test Suite 2.15.txt" which is part of the souce code and here are also interesting infos:

http://plus4world.powweb.com/software/Test_Suite

```
///////////////////////////////////////////////////////////////////////////////
Programs TRAP* - 6510 IO traps, page boundaries and wrap arounds

 #  code  data  zd  zptr   aspect tested
-----------------------------------------------------------------------------
 1  2800  29C0  F7  F7/F8  basic functionality
 2  2FFE  29C0  F7  F7/F8  4k boundary within 3 byte commands
 3  2FFF  29C0  F7  F7/F8  4k boundary within 2 and 3 byte commands
 4  D000  29C0  F7  F7/F8  IO traps for code fetch
 5  CFFE  29C0  F7  F7/F8  RAM/IO boundary within 3 byte commands
 6  CFFF  29C0  F7  F7/F8  RAM/IO boundary within 2 and 3 byte commands
 7  2800  D0C0  F7  F7/F8  IO traps for 16 bit data access
 8  2800  D000  F7  F7/F8  IO trap adjustment in ax, ay and iy addressing
 9  2800  29C0  02  F7/F8  wrap around in zx and zy addressing
10  2800  29C0  00  F7/F8  IO traps for 8 bit data access
11  2800  29C0  F7  02/03  wrap around in ix addressing
12  2800  29C0  F7  FF/00  wrap around and IO trap for pointer accesses
13  2800  0002  F7  F7/F8  64k wrap around in ax, ay and iy addressing
14  2800  0000  F7  F7/F8  64k wrap around plus IO trap
15  CFFF  D0C6  00  FF/00  1-14 all together as a stress test
16  FFFE  ----  --  --/--  64k boundary within 3 byte commands
17  FFFF  ----  --  --/--  64k boundary within 2 and 3 byte commands

In the programs TRAP16 and TRAP17, the locations of data, zerodata and zeroptr depend on the addressing mode. The CPU port at 00/01 is not able to hold all 256 bit combinations.

The datasette may not be connected while TRAP16 and TRAP17 are running!
```

The last sentence is very relevant:

Between V5 and V5.1 we added support for the real-time clock but we activated it hardwired. This means, "the datasette" (the real-time clock) is always connected in V5.1. Hypothesis: This might be the reason why the trap fails.

@MJoergen As discussed on Skype: Please do a testbuild where you disconnect the "datasette" and check if this fixes it.

MJoergen, 2024-03-14:
@paich64

The attached file contains V5.1 A16 for both R3 and R6 board. Please try both.

This version is based of the RC2, but with the RTC disabled.

[C64MEGA65_V51A16.zip](https://github.com/MJoergen/C64MEGA65/files/14601501/C64MEGA65_V51A16.zip)

MJoergen, 2024-03-14:
@paich64 A clarifying question: Does both TRAP16 and TRAP17 complete without any errors, on the MiSTer ? I'm specifically talking about the errors in TRAP17 that were already present in V5.0.

paich64, 2024-03-14:
@sy2002 @MJoergen 

Good news, the issue is fixed with V5.1A16 !

**Results comparison between V5.0 / V5.1RC1 / V5.1RC2 / V5.1A16 on R3A and R6 boards** 

**V5.0**   
R3A : TRAP1-16 : OK    TRAP17 : KO but does not hang

**V5.1RC1/RC2** 
R3A : TRAP1-15 : OK    TRAP16 : KO TRAP17 : KO and hangs 
R6  : TRAP1-15 : OK    TRAP16 : KO TRAP17 : KO and hangs 

**V5.1A16** 
R3A : TRAP1-16 : OK    TRAP17 : KO but does not hang
R6  : TRAP1-16 : OK    TRAP17 : KO but does not hang 

**Mister Latest core** 
TRAP1-16 : OK    TRAP17 : KO but does not hang

**Basically, on the Mister latest core, the test suite behaves exactly the same as on V5.0 and V5A16 on R3A/R6**

Let me know if i should keep using V5A16 for smoke tests.

sy2002, 2024-03-15:
@paich64 No, please do not use Alpha 16 for additional smoke tests or demo tests. Please do use RC2 instead.

Reason: Alpha 16 does nothing else than DISABLING the real time clock for GEOS (which useses the cassette port) because the test suite wants the cassette drive disconnected as described above: "The datasette may not be connected while TRAP16 and TRAP17 are running!" This is the only difference between Alpha 16 and RC2.

MiSTer seems to have disabled the RTC for GEOS by default.

For the time being, we want to hide this complexity from the end users of the MEGA65 and the GEOS RTC should "just work". Which is why for the time being we would tend to leave the RTC on by default - which means TRAP 16 would continue to fail. 

But if thorough tests such as smoke tests, cartridges (hardware or emulated) or demos would fail because of this, then we would reconsider.

PLUS: It might make sense to have a "Advanced settings" submenu in V6 where users might be able to configure things like this, but also other things like VIC models, CIA models as described here: https://github.com/MJoergen/C64MEGA65/issues/120

I am leaving this issue open so that we can re-consider at a later stage.

paich64, 2026-09-02:
@sy2002 I confirm that with **A20X2** the issue is fixed : 

Not only the issue can be reproduced by enabling RTC clock in advanced settings, but also,
The issue is gone as soon as RTC clock is disabled in advanced settings.

From a GEOS perspective, funny enough I had never understood you could take advantage of the RTC chip which is inside the Mega65. I had always seen the clock ticking in GEOS but i had never figured out you could use the internal RTC chip :) 

So I included the needed driver in the main GEOS system disk and tested the real time clock : 

When RTC clock is disabled in advanced settings, indeed it's behaving as it used before (ticking but not synced with my Mega65 RTC)
When RTC clock is enabled in advanced settings, now for the very first time I have Geos running with its clock synced with the Mega65 RTC !

So you can close this issue.

closed 2026-09-13 by sy2002