MJoergen/C64MEGA65 #135: Demo "Old Men in Used Cars" by "Fossil" stops working with V5.1RC2

paich64, 2024-04-28:
Demo "Old Men in Used Cars" by "Fossil" stops working with V5.1RC2 : 

When inserting disk n°2, the first picture gets loaded and the music plays, but the visual effects just stop.

Since there is an already exisiting issue with V5.1RC2 https://github.com/MJoergen/C64MEGA65/issues/133 I have tried the demo with the V5.1A16 core which fixes issue 133, and the demo runs perfectly as with V5.0.

sy2002, 2024-05-17:
My hypothesis is, that this regression was introduced by the GEOS RTC feature that has been introduced in V5.1. A strong data point that backs this hypothesis is, that V5.1A16 resolves the issue: Alpha 16 was a never-released Alpha version that did nothing else than deactivating the RTC that uses the C64's tape port. So it looks like this demo does not like if the tape port is active while it runs, just like TRAP16 in the emulator test suite: https://github.com/MJoergen/C64MEGA65/issues/133

The decision is to live with this regression in V5.1 as the benefit of having a RTC for GEOS is larger than this singular demo. In future, we will have advanced compatibility settings and will be able to switch the RTC on/off. Today the default is ON and in future it might be OFF just to be more compatible.

https://github.com/MJoergen/C64MEGA65/issues/133

sy2002, 2025-04-27:
@paich64 You might want to confirm with V5.2X1 if this fixes it.

paich64, 2026-09-02:
@sy2002 I confirm the issue is fixed on VA20X2

- When Geos RTC is enabled in advanced settings, the issue still occures.
- When Geos RTC is disabled in advanced settings, the issue is gone : we at last get the expected effect while music is playing and next demo parts run as expected.

You can close this issue.

closed 2026-09-13 by sy2002