MJoergen/C64MEGA65 #136: Allow simulated REU in parallel to using hardware cartridges

sy2002, 2024-05-02:
There is a thread on Discord, initiated by AmokPhaze101 aka @paich64 about if and how and why we should support the simulated REU in parallel to using hardware cartridges; click here to read the full thread:

https://discord.com/channels/719326990221574164/794775503818588200/1235344320811372584

Summary:

* There are interesting use cases when it comes to software development and debugging: "The Snappy ROM which can be flashed to EasyFlash 3 cartridge (provided the EF3 is programmed with a specific CPLD) contains a version of Turbo Macro Pro with REU support." and "EF3 supports different freezers and snappyrom is an improved replacement rom for Super Snapshot V5. It also embeds the turbo Macro Pro version which can either use C64 ram or use REU (if REU is detected) so that the dev gets all the C64 ram memory available for his program. With such a combination of Freezer/Macro Assembler + REU support we have the ultimate dev/debug tool.  https://github.com/adrianglz64/snappyrom"

* Games, such a "A Pig Quest" (hardware cartridge) can make use of REU: "Same applies for "A pig Quest" hardware cartridge which supports REU to accelerate the loading of levels. It's not possible to use this feature on the C64 core."

* For real Commodore 64 hardware "there are cartridge expanders in which you can plug more than one cartridge which can work together. Typically you can have A cartridge Datel Midi Interface working with the Cynthcart 64 cartridge, both plugged in the Cartridge expander."
*

sy2002, 2024-10-30:
[image]

MJoergen, 2026-04-25:
Should be fixed in V6.0 Alpha Release 9