MJoergen/C64MEGA65 #138: Wonderboy CRT does not work

sy2002, 2024-06-13:
The CRT version of Wonderboy does not work but it works fine with Vice:

[Wonderboy_1987Activision.crt.zip](https://github.com/user-attachments/files/15812950/Wonderboy_1987Activision.crt.zip)

paich64, 2024-06-13:
Indeed, I checked and the "OCEAN TYPE 1" (5) version which used to work, does not work anymore.
However the attached "Magic Desk, Domark, HES Australia" (19) version keeps working fine.

I have tested other OCEAN TYPE 1(5) cartridges and they also work fine.

[Wonderboy_MagicDesk.zip](https://github.com/user-attachments/files/15814830/Wonderboy_MagicDesk.zip)

sy2002, 2024-06-13:
@paich64 Can you confirm, that it did work with V5.0?
P.S. woepaul did send me a private DM to report this.

paich64, 2024-06-13:
@sy2002 if you check https://github.com/MJoergen/C64MEGA65/blob/master/tests/carts.md you can see that Wonderboy in "Ocean" (5) format was not part of my tests.
Actually i had tested the MagicDesk Version (which is the same game but using a different cartridge format).

I have just tested the "Ocean"(5) version of Wonderboy and i confirm it does not work on V5.0. 

As far as i know there are only 2 versions : The MagicDesk one (which is working) and the Ocean(5) one (which is not working).

Beside a possible fix for the Ocean version, I guess woepaul can use the MagicDesk version.

sy2002, 2024-06-14:
Thank you Olivier, this is great feedback, because it shows, that this is NOT A REGRESSION of V5.1 and therefore for sure no release blocker. It is "just" a bug in our Ocean Type 1 implementation that we can tackle at a later stage.

MJoergen, 2024-11-10:
I've disassembled the beginning of the file Wonderboy_1987Activision.crt. It starts at 0x8009:

```
8009   8E 16 D0             STX $D016
800C   20 A3 FD             JSR $FDA3
800F   20 50 FD             JSR $FD50
8012   20 15 FD             JSR $FD15
8015   20 5B FF             JSR $FF5B
8018   58                   CLI
8019   20 53 E4             JSR $E453
801C   20 BF E3             JSR $E3BF
801F   20 22 E4             JSR $E422
```

The last line prints the BASIC start up messages, and particularly calls into the function at 0xA408. So, this cartridge apparently expects the BASIC ROM to be mapped into 0xAxxx, i.e. it considers the cartridge to be an 8 kB type (only ROML).

However, the source code in VICE has the following comment:
**- the hardware uses 16k game config, the same 8k ROM bank is mapped to both ROML and ROMH.**
There seems to be a contradiction here.

MJoergen, 2024-11-11:
Well, I'm not completely wrong.

I've verified the cartridge works in vice 3.8, and from the debug log I see:

```
...
 462518 .A7F2 310 002    1650980  FF FF FF    ISB $FFFF,X 010008fb
 462519 .A7F5 310 009    1650987  FF FF FF    ISB $FFFF,X 010008fb
 462520 .A7F8 310 016    1650994  FF FF FF    ISB $FFFF,X 010008fb
 462521 .A7FB 310 023    1651001  FF FF FF    ISB $FFFF,X 010008fb
 462522 .A7FE 310 030    1651008  FF FF 00    ISB $00FF,X 010008fb
 462523 .A801 310 037    1651015  0B 08       ANC #$08   000008fb
 462524 .A803 310 039    1651017  C4 07       CPY $07    000008fb
 462525 .A805 310 042    1651020  9E 32 30    SHX $3032,Y 000008fb
 462526 .A808 310 047    1651025  37 38       RLA $38,X  000008fb
 462527 .A80A 310 053    1651031  46 49       LSR $49    000008fb
 462528 .A80C 310 058    1651036  4C 45 20    JMP $2045  000008fb
 462529 .2045 310 061    1651039  FF 00 00    ISB $0000,X 000008fb
 462530 .2048 311 005    1651046  00          BRK        cf0008fb
 462531 *** BRK
 462532 .FF48 311 012    1651053  48          PHA        cf0008f8
 462533 .FF49 311 015    1651056  8A          TXA        cf0008f7
...
```

So it seems that the CPU does indeed execute invalid instructions ...

MJoergen, 2024-11-12:
This is very similar to issue #69 .

sy2002, 2024-11-12:
> Apparently, the expected behavior is to **mirror** the data from ROML to ROMH in that case.

Sounds good - just being curious: How does one know, when to mirror?

MJoergen, 2024-11-12:
> > Apparently, the expected behavior is to **mirror** the data from ROML to ROMH in that case.
> 
> Sounds good - just being curious: How does one know, when to mirror?

Yes, how does one know when to mirror? ... Well, for starters it depends on cartridge type. Other than that, basically I rely
on the source code of VICE, e.g. here: https://github.com/VICE-Team/svn-mirror/blob/main/vice/src/c64/cart/ocean.c#L73-L74 .

MJoergen, 2024-11-12:
Here's why the banks are mirrored: ROML and ROMH are tied together. This is part of the schematic for the cartridge:
[image]
The above image is taken from https://www.zimmers.net/anonftp/pub/cbm/schematics/cartridges/c64/freezer/MK.gif

closed 2026-04-22 by sy2002