MJoergen/C64MEGA65 #139: Add support for the Ultimate II+ cartridge

sy2002, 2024-06-14:
Details: https://ultimate64.com/

Here is a snapshot of a discussion that happened on Discord:

[image]

sy2002, 2024-06-14:
@wiebow please follow this issue and become a participant.

sy2002, 2024-06-15:
@CINJ as discussed on Discord, please follow this issue and become a participant

sy2002, 2024-12-08:
@paich64 Can you please check, if the V5.3A2 fix fixes this?

paich64, 2024-12-09:
No it does not work with V5.3A2 

- the autoboot feature won't launch the C64 core with the Ultimate cartridge plugged it.
- Manually forcing the C64 core V5.3A2 results in the core booting, but hitting the middle button to have the Ultimate cartridge OSD showing up, results in the C64 core V5.3A2 displaying strange characters or different background colour. 
- the Ultimate Cartridge Reset, though, resets the C64 core.

I will have to plug an real C64 and check how my OSD menu is setup, it may just be set up to display the inlaid OSD instead of the full screen OSD. Maybe that makes a difference. Will check this week-end when i have a chance tu plug back my real C64, and check how the cartridge OSD menu is setup.