MJoergen/C64MEGA65 #142: Use F9/F11/F13 for Core Shortcuts

Kugelblitz360, 2024-06-17:
As F15/Help already is captured and processed by the Core, giving F9/F11/F13 some (user-definable?) extra love would be awesome. The easiest and most useful shortcuts I would see would be "Mount Disk, Start PRG, Insert Cart" for the three buttons.

Once mount lists become a feature, "Next Disk, Previous Disk, Mount Disk" would be a great alternative use.

Anything else I guess is not used so often so can be on the HELP menu, but if the code to go to ANY function in the menu can be added, users could set these three F-Keys to their own liking.

sy2002, 2024-06-19:
Thank you @Kugelblitz360 . Great idea.

sy2002, 2026-11-10:
@Kugelblitz360 - Adding this to the V6 scope.

sy2002, 2026-11-14:
@Kugelblitz360 Please test [C64MEGA65-WIP-V6-A22X3.zip](https://github.com/user-attachments/files/34850669/C64MEGA65-WIP-V6-A22X3.zip) and confirm here if the shortcuts work as intended.