MJoergen/C64MEGA65 #143: TeensyROM is not booting its menu

sy2002, 2024-06-23:
@dansanderson reported via Discord:

[image]

I do not know the TeensyROM cartridge, yet and assume it is this one:

https://github.com/SensoriumEmbedded/TeensyROM

@dansanderson please clarify/confirm: I do assume you tested on an R6 board using V5.1 RC3?

dansanderson, 2024-06-24:
Correct, that's TeensyROM. C64 core WIP-V5.1-RC2.

Technically we tested on an unpatched R5, which might make a difference. I didn't get around to installing the bodge wires that IIRC we told Trenz to install on the R6. I should have an R6 to test with when my number comes up soon.

sy2002, 2024-06-24:
@dansanderson Indeed it would be great if you could confirm that the problem persists on a R6 board. Reason: The C64 core explicitly makes use of "hardware stuff" that requires a modded R5 or an R6 board. So the reason for this bug MIGHT be that your R5 board is not modded. (But I am not claiming this is the reason ;-) It MIGHT be.)

paich64, 2024-08-03:
I just got my Tensyrom and tested it with my R6 board. 

Here are the observations using the c64 core V5.1 : 

1) When booting the Mega65 with the Tensyrom inserted, the C64 core is started, meaning, the cartridge is detected as a C64 cartridge 
2) However, the TensyROM menu won't show show up
3) Pressing the "main menu" TensyROM button will in core being reset (obviously) but the menu won't show up 
4) Short/long reset of the C64 core won't help 

Here are the observations when enforcing the MEGA65 core to boot with the cartridge inserted (Holding NOSCROLL and selecting the MEGA65 core in slot 1) 

=== specific to the MEGA65 core ===
1) Most of the of the time, the TensyROM main menu will show up 
2) When the TensyROM menu does not show up (instead we have the Mega65 basic prompt) a simple reset, or using GO64 will result in the TensyROM menu showing up
3) Contrary to the C64 core, the "main menu" TensyROM button does not trigger any reset (while it should) 
4) .prg will load and work as expected 
5) .crt will keep loading forever and will never start
6) F2 which is supposed to leave the menu and go back to the Basic prompt does not work and result in freezing all other keys which are usually working  
=========================

The above observations are only related to the Mega65 core. Since the menu does not show up using the C64 core, i can't test further with it. 

I have ordered a second TENSYROM and will ship it to @MJoergen as soon as i get it.
I confirm it's this cartridge : https://github.com/SensoriumEmbedded/TeensyROM
I confirm that TRAVIS agrees to help us with the troubleshooting.

paich64, 2024-09-14:
update : Tensyrom will be shipped to Michael next monday. In the meantime here is the behaviour observed on 3 different configurations : 

**TENSYROM on Ultimate 64**  This is the expected behaviour.

https://www.youtube.com/watch?v=zu5A68Ha8nY

switching on the Ultimate 64 results in the cartridge booting and starting the tensyrom menu. 
we can hit F3, browse the Tensyrom SD card, select a .prg file, the .prg file gets loaded and the program started.
Hitting the Tensyrom button reset button results in the tensyrom cartridge being reset (loading again the the tensyrom menu).
Holding the reset button a long time results in the catrige starting its reset procedure but being frozen (as if cpu was stopped) and releasing it
will result in the tensyrom cartridge reset procedure completing (loading again the the tensyrom menu).

**TENSYROM with the MEGA65 core 0.96**

https://www.youtube.com/watch?v=tLNC_dhvpHY

the c64 core is not installed, so if a c64 cartridge is inserted the Mega65 core will start in GO64 mode.

switching on the Mega65 WITHOUT the C64 core installed and the Tensyrom inserted results in the GO64 mode starting and the cartridge booting and starting 
the tensyrom menu. As on a real C64, we can hit F3, browse the Tensyrom SD card, select a .prg file, the .prg file gets loaded and the program started.
Note : because of the limitations of the GO64 mode, not all .prg will load and play.

The tensyrom reset button does not work when in GO64 mode : pressing it does nothing.
To go back to the Tensyrom menu, we have to push the Mega65 reset button.

**TENSYROM with the C64 core V5.1**

https://www.youtube.com/watch?v=U397i7pEvgc

C64 core installed and configured to start when a c64 cartridge is inserted.

switching on the Mega65 with the C64 core installed and the Tensyrom inserted results in the C64 core starting.
This means that the tensyrom is detected as a C64 cartridge. However, The tensyrom menu does not show up.

The video show the Mega65 booting the C64 core (but no tensyrom menu shows up). Then 2 short resets are consecutively 
executed using the Tensyrom reset button : each time a black screen appears and we get back to the C64 "ready" prompt. 
Then 1 long reset is executed with the Tensyrom reset button (we keep the tensyrom reset button held a few seconds before releasing it) :
The black screen appears and stays until we release the Tensyrom reset button. When we release it we get back to the C64 "ready" prompt.

Kugelblitz360, 2024-09-14:
And @sensoriumembedded mentioned to me that he is holding the Reset line for approx. 50.000 Phi2 cycles in order to initialize itself; this halts the CPU.

SensoriumEmbedded, 2024-09-14:
The 50mS reset hold time was chosen somewhat arbitrarily.  If you think holding it that long is the cause, I'm happy to do a build with shorter reset time to help prove it out.  Just let me know, thanks!

Kugelblitz360, 2024-10-11:
When the MEGA65 is in M65 Core mode, GO64 activates the (physical) TeensyROM cartridge. It boots, Music plays, Menu works. Internal functionality like updating via SD card works too.

In the C64 Core the cartridge does not boot but the RAM test in Reset sees that there is a ROM and outputs 30719 BASIC BYTES FREE

A quick and dirty test:

10 PRINT PEEK (32768)
20 GOTO 10
show the value 255 with a 70% chance. In the other 30% of cases it shows either 9 (the correct value for most carts), 63 or 127.
Doing the test with 32769, which most often contains $80, gets again most often 255, some 128, some 191 and some 159.

So somehow the cart simply is not read correctly.

_In the setup screen (f8) "RW Read Dly" is default "Off".  You could set this to "On" (it gets saved in EEPROM) then power off and back on to apply it and see if you notice anything different._

When this is done, the values switch to 9, 41 and 57. Getting there, 9 is the correct value, but still there is something wrong.

sy2002, 2024-10-11:
@Kugelblitz360 Can you please check what happens, if you turn "HDMI: Flicker-free" OFF, save the settings (confirm by resetting without the cartridge) and then try again?

Kugelblitz360, 2024-10-11:
POKE 32768,0 or POKE 32768,255 does not change anything, so it is not that it sometimes reads the RAM. 

Sanity checked with 32767 too, there PEEK gives the value I poked.

SensoriumEmbedded, 2024-12-04:
I've documented my findings around GO64 and C64 Core compatibilities with the TeensyROM in the attached documents.  This includes recommendations for code improvements on the Mega65 side.
I've temporarily worked around the largest issue with TR firmware to get it working in C64 core mode.   [FW available here](https://github.com/SensoriumEmbedded/TeensyROM/tree/main/bin/TeensyROM/Mega65) 

Findings docs:
[Mega65+TeensyROM using C64 Core.pdf](https://github.com/user-attachments/files/18002196/Mega65%2BTeensyROM.using.C64.Core.pdf)
[Mega65+TeensyROM using GO64.pdf](https://github.com/user-attachments/files/18002198/Mega65%2BTeensyROM.using.GO64.pdf)

MJoergen, 2024-12-06:
Here is a screenshot from the FPGA's built-in logic analyzer (ILA):

[image]

The numbers at the top are core clock cycles @ 32 MHz, i.e. the units are approx 32 ns.

The interesting traces are `cart_a_o` (address output) and `cart_d_i` (data input), as well as `cart_phi2_o`. Here the trace is reading from address 0x8008, and the correct value 0x30 is available already after 8 clock cycles (250 ns) later. However, when `cart_phi2_o` has the falling edge, the data input changes to 0x7F (not visible in the picture). This is the value seen by the CPU, so that indicates that the CPU is sampling the data lines right around the falling edge of the PHI2 clock.

I've made a small fix in commit 03960c66 (on the develop branch) that delays the PHI2 clock to the cartridge by one clock cycles (i.e. 32 ns). With that fix the TR menu appears.

However, since this fix changes the timing on the cartridge port, we need to perform an extensive test of the existing hardware cartridges to check for any negative side effects of this fix (i.e. regressions). Therefore, I've put this fix on the develop branch, which is the target for the later V5.3 release.

However, if anyone is willing to give this fix a spin, please do and please report back here with your findings.

Kugelblitz360, 2024-12-06:
If you would build that core, we could test Sidekick64 against it which also fails to run currently. Would you mind sharing a binary, maybe on our other communication channel?

Screenshot.from.2024-12-06.10-10-04.png (view on web)<https://github.com/user-attachments/assets/57e19271-3253-4d68-89c7-868a076060e6>

I've made a small fix in commit 03960c6<https://github.com/MJoergen/C64MEGA65/commit/03960c66d14862393e65e463f790d38e31b2a143> (on the develop branch) that delays the PHI2 clock to the cartridge by one clock cycles (i.e. 32 ns). With that fix the TR menu appears.

paich64, 2024-12-06:
Hi just made a quick regression test with V5.3A1 and my hardware cartridges :   

They are all working with V5.1 Final 

With V5.3A1 :    

Zetawings                                   => won't boot
EF1 with "Eye of the beyholder" => won't boot 
Robot Ject Action                       => won't boot 
Super BreadBox                          => won't boot 
Protovision Sam's Journey          => won't boot 
Protovision Galencia                   => Won't boot 

EF3                                                                             => boots, will boot any game, especially "Eye of the beyholder" and works
EF1 with "A pig Quest"                                              => boots and works
Protovision "Soulforce"                                             => boots and works 
Protovision "A pig quest"                                          => boots and works 
Seawolves                                                                  => boots and works
Avengers                                                                   => boots and works 
Pinball Spectacular                                                    => boots and works 
Hesware "Grid Runner"                                             => boots and works 
Custom made cartridge "Creatures"                         => boots and works 
Custom made cartridge "Mayhem in Monsterland" => boots and works 
Power Cartridge                                                        => boots and works 
Custom made Super Snapshot V5                            => boots and works 

Interesting to note that it's not all EF1 which fail : It looks like it depends on the game which is flashed on EF1.

paich64, 2024-12-07:
Adding the following 2 cartridges which keep working as expected with V5.3A2 👍 

Action Replay VI => boots and works
Final Cartridge III => boots and works

paich64, 2024-12-07:
Tested the TensyROM supported cartridge types with the C64 core V5.3A2 (Lastest Tensyrom firmware) 

All the supported type are working perfectly when launched from Tensyrom.

Cart_Generic                   0  ===============> OK , included Ultimax Games !
Cart_Oceantype1            5  ===============> OK
Cart_FunPlayPowerPlay  7  ===============> OK
Cart_SuperGames           8  ===============> OK
Cart_EpyxFastload         10  ==============> OK
Cart_C64GameSystem3 15  ===============> OK
Cart_Dinamic                 17  ===============> OK
Cart_ZaxxonSuper         18  ===============> OK
Cart_MagicDesk            19  ===============> OK
Cart_EasyFlash               32  (file size limit and no EAPI support) ===============> OK

paich64, 2024-12-07:
After having tested many .PRG files which currently do not work with the C64 core (to be fixed in 5.2) I confirm .PRG loading works nicely when loaded via TENSYROM using V5.3A2.

sy2002, 2024-12-08:
@MJoergen I checked the commit and delaying the PHI2 output is indeed a "dangerous" change as it changes absolutely everything we do on this hardware port. On the other hand: @paich64 's tests sound EXTREMELY promising. I reviewed your fix and it is clean and easy. So I would tend to even add it to **V5.2** if you agree and if @paich64 can confirm that...

a) ... these encouraging results are working on R3 and R6 systems

b) ... **FLASHING the EF1 and EF3** still works like a charme on R3 and R6 systems: Flash the EasyFlash 1CR with a small (<64k) and large (>512k) game and playtest these games, Flash the EasyFlash 3 with a small (<64k) and large (>512k) game and playtest these games

c) ... saving and loading savegames still works on R3 and R6 systems (example Sam's Journey save game, load game)

paich64, 2024-12-08:
@sy2002 @MJoergen remember that my R3A is not reliable at all when it comes to the cartridge port ....
So I'm afraid but when it specifically comes to the cartridge port i can't provide meaningfull results.
Just for the record, EF1 do not work anymore on my R3A, KFF has never worked on my R3A, even with the workaround ......

I will confirm b) and c) for R6 at least.

For R3A, despite the non reliability of my very own cartridge port, i only have the R6 version of V5.3A2.

Kugelblitz360, 2024-12-08:
I will be testing TeensyROM network, NFC and other USB functionality (not MIDI I am afraid, but if I can get network to work), then Sidekick64 (excluding the 2nd SID emulation because that needs extra wires to the (non existing) SID inside the C64), then EF3 flashing and game saving and many more KFF tests with writing to the SD Card.

Does the timing change positively affect the skoe Kernal replacement trick? https://skoe.de/kernal/[kernal-cartridge.pdf](https://skoe.de/kernal/kernal-cartridge.pdf)

MJoergen, 2024-12-08:
> @MJoergen I checked the commit and delaying the PHI2 output is indeed a "dangerous" change as it changes absolutely everything we do on this hardware port. On the other hand: @paich64 's tests sound [...]

Historically, the V5.2 release was meant to fix the "HDMI jailbar issue" quickly, and to avoid abusing @paich64 's time and willingness to test. The V5.3 release is looking to become a "cartridge" release, fixing bugs in both physical and simulated cartridges. I even plan to looking into supporting simultaneously physical cartridge, simulated cartridges, and the simulated REU. The use case is a simulated cartridge with a game that needs REU, with an attached physical freezer cartridge. I can't find a GitHub issue with this feature request, but I think it should be doable. This seems like a nice feature to include in the V5.3 release.

So to summarize, I think we should avoid "feature creep" in the V5.2 release, and keep it simple.

MJoergen, 2024-12-08:
> Does the timing change positively affect the skoe Kernal replacement trick? [https://skoe.de/kernal/[kernal-cartridge.pdf](https://skoe.de/kernal/kernal-cartridge.pdf)](https://skoe.de/kernal/%5Bkernal-cartridge.pdf%5D(https://skoe.de/kernal/kernal-cartridge.pdf))

It's quite likely to have an impact. I would need to study this document in detail, so not likely to happen right now. Is there already a GitHub issue with this bug ?

Kugelblitz360, 2024-12-08:
It's the way Easyflash 3 does Kernal replacement, which you actively switched off because so far it did not work.

sy2002, 2024-12-08:
@MJoergen OK agreed, then this one stays in V5.3 as it might become a "cartridge release". The smaller PRs from csoren go to V5.2 as discussed there.

@MJoergen EF3 kernal replacement will not work without larger tricks (particularly on R3 boards) as the EF3 does electrically some things that work on a C64 but that might overload the input/output drivers of the MEGA65 PCB. More details are here: https://github.com/MJoergen/C64MEGA65/issues/126   Summary @Kugelblitz360: This fix has nothing to do with the EF3 kernal replacements.

@Kugelblitz360 and @paich64 Thank you for testing what I have written above when time is ripe. Please document here.

Kugelblitz360, 2024-12-09:
Very first tests show that the 5.3A2 changes made Sidekick64 (https://github.com/frntc/Sidekick64) boot and work. Stress tests to follow. But to me that change looks more and more promising because it enables more of the Software-Defined carts which in theory should be more critical than the good old EPROM based ones...

sy2002, 2026-02-05:
@Kugelblitz360 Thank you for this very valuable update and info. 

@MJoergen for your convenience: The DMA line from the C64's expansion port is an input-only signal - see screenshot from C64 Wiki:

[image]

We are not using this signal, it is just assigned here and never used:

https://github.com/MJoergen/C64MEGA65/blob/master/CORE/vhdl/main.vhd#L865C10-L865C20

The actual MiSter core does expose the DMA signal and you @MJoergen used it when implementing the REU. Just look here and search for "core_dma" in the whole file:

https://github.com/MJoergen/C64MEGA65/blob/master/CORE/vhdl/main.vhd#L690C25-L690C33

So fixing this might be as easy as wiring it here:

https://github.com/MJoergen/C64MEGA65/blob/master/CORE/vhdl/main.vhd#L906

And double-checking three/four other places in main.vhd.

Note to myself when writing release notes and note to @Kugelblitz360 when updating the documentation: As far as I know this only works on R6 (and later) machines, i.e. we will need to document this properly.

Kugelblitz360, 2026-02-26:
Version 6.0 Alpha 2 passes the TeensyROM DMA tests!