MJoergen/C64MEGA65 #144: "Quite a few" .PRG files are not working

sy2002, 2024-06-30:
### Here is the problem:

* "Quite a few" .PRG files are not working
* This is not a regression of 5.1: The problem already existed on 5.0

### Some details:

Reported by Forum64 user clarkkent: https://www.forum64.de/index.php?thread/125676-c64-core-f%C3%BCr-den-mega65/&postID=2159973#post2159973

And also via dm on Discord:

[image]

### Questions for Clark Kent:

* Can you name a few .PRGs from your private repository that do not work (just list the names of the .prg files here)
* Please check: What is the setting of the "Expansion Port" menu while you encounter this problem: Is it "Use hardware slot" (and do you have an attached cartridge?), is it "Simulate 1750 REU" or is it "simulate cartridge" (and do you have a simulated cartridge up and running)?
* .PRGs that don't work: Is the also happening after cold starting the core (power-on)? Is this also happening after you made a [hard-reset/long-reset](https://github.com/MJoergen/C64MEGA65?tab=readme-ov-file#hard-reset-vs-soft-reset)
* Or is it maybe only happening after you loaded a .prg and then the next and then the next?
* .PRGs that don't work: Are they always failing or only in a certain context like "first load .prg A and then .prg B fails, but if I load .prg B first after a cold start, THEN it works fine"

cwscws, 2024-06-30:
First: Yes, I did everything that it recommended. Usually "Simulate Cartridge" is selected but I also tried the other options without any success. And it's also after a cold start or long-reset. And there is not combination or way to make it work - the PRGs that are not working are not working in any case. It also does not matter in which order I start the PRGs. I also tried several other settings in the menu, no difference.

Here a short list (I tried about 15 to 20 games, these had the problems):
ALLEYKAT.PRG (no reaction after RUN)
ANTIRIAD.PRG (no reaction after RUN)
ARKANOID.PRG (no reaction after RUN)
CHOPLIFTER.PRG (no reaction after RUN)
HIGHNOON.PRG (RUN -> READY.)
MR. DO!.PRG (RUN - > READY.)

All of these games are working perfectly in any C64 emulator.

paich64, 2024-06-30:
> First: Yes, I did everything that it recommended. Usually "Simulate Cartridge" is selected but I also tried the other options without any success. And it's also after a cold start or long-reset. And there is not combination or way to make it work - the PRGs that are not working are not working in any case. It also does not matter in which order I start the PRGs. I also tried several other settings in the menu, no difference.
> 
> Here a short list (I tried about 40 - 50 games, these had the problems): ALLEYKAT.PRG (no reaction after RUN) ANTIRIAD.PRG (no reaction after RUN) ARKANOID.PRG (no reaction after RUN) CHOPLIFTER.PRG (no reaction after RUN) HIGHNOON.PRG (RUN -> READY.) MR. DO!.PRG (RUN - > READY.) 1943.PRG (RUN -> stuck in light blue screen) APPLE CIDER SPIDER.PRG (no reaction after RUN) BOMB JACK.PRG (destroyed graphics right after RUN) BURGER TIME.PRG (no reaction after RUN) ENIGMA FORCE.PRG (no reaction after RUN) POOYAN.PRG (no reaction after RUN) PITFALL.PRG (characters on screen like from a decruncher, then no reaction) TRAILBLAZER.PRG (RUN -> READY.) SQUISH.PRG (no reaction after RUN) SPINDIZZY.PRG (RUN -> READY.) ZORRO.PRG (RUN -> READY.) YIE AR KUNG FU.PRG (RUN -> READY.)
> 
> All of these games are working perfectly in any C64 emulator. Also no problems on the MiSTer with these games - all are working.

I confirm the issue on the c64 core with Antiriad, and yes it works on the Mister. It may be related to .prg injection since as a workaround i have put ANTIRIAD.PRG on a blank .d64 (created with DIRMASTER) and the game will properly load in the C64core. So while it's not a fix, it's a workaround for the moment.

paich64, 2024-06-30:
Here is an archive with the .prg which does not work, and the .d64 using the exact same .prg and which is working as expected.
[antiriad.zip](https://github.com/user-attachments/files/16045527/antiriad.zip)

paich64, 2024-06-30:
@MJoergen @sy2002 Let me know if you want me to test all these releases (as well as the workaround used for Antiriad) and i will.

paich64, 2024-06-30:
@MJoergen @sy2002 @cwscws Here is another version of Antiriad.prg (which obviously comes with some kind of fastloader) and it's running correctly on the C64 core. 
[Antiriad (1986)(Palace Software).zip](https://github.com/user-attachments/files/16045564/Antiriad.1986.Palace.Software.zip)

paich64, 2024-06-30:
@cwscws  Any chance you share here a .zip file containing the .prg files you have identified as "not working on the C64 core" ? I would like to do further tests / investigation being sure i'm working with the exact same .prg you are using. Thanks.

MJoergen, 2024-06-30:
@sy2002 
I believe the problem is our `prg_loader.vhd` file.

Here are my findings so far:
In the game `antiriad.prg` I've disassembled the first few lines of code. They are:
```
0810   78         L0810     SEI
0811   A2 FF                LDX #$FF
0813   9A                   TXS
0814   A0 00                LDY #$00
0816   84 01                STY $01
0818   A2 CC                LDX #$CC
081A   BD 57 08   L081A     LDA $0857,X
081D   9D 33 03             STA $0333,X
0820   B9 1D 09             LDA $091D,Y
0823   99 FF 00             STA $00FF,Y
0826   CA                   DEX
0827   D0 02                BNE L082B
0829   A2 01                LDX #$01
082B   88         L082B     DEY
082C   D0 EC                BNE L081A
082E   C6 2E      L082E     DEC $2E
0830   A5 2E                LDA $2E
```
What is troubling to me are the last two lines, accessing the zero page variable $2E. In the C64 memory map I find the following:
```
$002D-$002E : Pointer to beginning of variable area. (End of program plus 1.)
```
I don't see anywhere in our `prg_loader.vhd` that we set this variable.

Now, this is not the only variable needed. If you look in the MiSTer file [c64.sv](https://github.com/MJoergen/C64_MiSTerMEGA65/blob/5afb90e02fcefb053ac27792f42e742ad61e8b9b/c64.sv) lines 740-774 we see a list of zero-page variables that must be set:
```
$002B-$002C : Pointer to beginning of BASIC area.
$00AC-$00AD : Start address for SAVE to serial bus.
$002D-$002E : Pointer to beginning of variable area. (End of program plus 1.)
$002F-$0030 : Pointer to beginning of array variable area.
$0031-$0032 : Pointer to end of array variable area.
$00AE-$00AF : Load address read from input file and pointer to current byte during LOAD/VERIFY from serial bus.
```

Note. I actually had the same problem for the VIC20 core, and this is how I implemented it there: https://github.com/MJoergen/VIC20MEGA65/blob/develop/CORE/vhdl/prg_loader.vhd

sy2002, 2024-07-01:
@MJoergen Excellent catch: When implementing the PRG loader back in the days, I did not realize that the C64's Zero Page "registers" need to be set properly - I ignored them. It is quite a miracle that it worked so often :-)   Thank you - I will take care.

@paich64 No need to test more PRGs at this moment in time. (And I do have @cwscws 's ZIP which I can share with you to test as soon as I fixed this).

paich64, 2024-07-01:
> @MJoergen Excellent catch: When implementing the PRG loader back in the days, I did not realize that the C64's Zero Page "registers" need to be set properly - I ignored them. It is quite a miracle [...]

Actually i just want to check if as a workaround, i can use a compresser to make these .prg working (again as a workaround) as it looks like the issue does not occure with .prg which use a compresser (possibly not using the same ZP). That would be i think an acceptable workaround if this works.

paich64, 2024-07-01:
> @MJoergen Excellent catch: When implementing the PRG loader back in the days, I did not realize that the C64's Zero Page "registers" need to be set properly - I ignored them. It is quite a miracle [...]

1) just to be exhaustive : HIGHNOON.PRG does not work in VICE
2) For the 17 other .PRG I have found this alternative if people want to be able to load the non loading .PRG fast : 

- They can get the following program : https://csdb.dk/release/?id=109269
- Then they make their very own custom simulated cartridge by adding all the .prg 
- Then they start the simulated cartridge with the C64 core : 
[image]

All the games will start.

cwscws, 2024-07-05:
Funnily enough, Q*bert is working now🤣

tonedef71, 2024-07-24:
Just curious.  When loading a `.PRG` file, how does the C64 Core know where in memory the binary contents of a file should be loaded?  Is it always the same memory location, or are starting memory locations encoded in the `.PRG` file itself?

MJoergen, 2024-07-24:
> Just curious. When loading a `.PRG` file, how does the C64 Core know where in memory the binary contents of a file should be loaded? Is it always the same memory location, or are starting memory locations encoded in the `.PRG` file itself?

The first two bytes of the PRG file contains the loading address.

MJoergen, 2025-03-16:
Should be fixed in V5.2A2

sy2002, 2025-03-29:
@paich64 Can you scroll up and test the PRGs originally reported by @cwscws with V5.2A2 and confirm that they work so that I can close this one?

paich64, 2025-03-31:
@sy2002 I never got the zip archive with the failing .prg files. The only one i got is antiriad, which now works fine with 5.2A2.

All the other games (which i also have in .prg format) have always run on 5.1 (possibly because they are coming from a specific collection where the .prg have been reworked to always start from the same default address which was ok with 5.1).

To me since antiriad is now working properly, i consider the issue as being fixed.

If you want me to test more, please provide the zip archive having all these faulty .prg (again the versions I have are perfectly working with 5.1, and that's why i suspect we did not spot the issue originally).

sy2002, 2025-04-07:
@cwscws This one is fixed with the upcoming V5.2 release. Thank you for reporting this.

closed 2025-04-07 by sy2002