MJoergen/C64MEGA65 #145: Vertical bars over HDMI

sy2002, 2024-06-30:
@paich64 @AnttiLukats Presumably our "favorite HyperRAM issue" is back:

User [TheOldMan on Forum64 reported](https://www.forum64.de/index.php?thread/125676-c64-core-f%C3%BCr-den-mega65/&postID=2160047#post2160047) these vertical bars over HDMI and @sho3string reported vertical bars in Arcade cores - I assume similar to these, @sho3string please confirm:

[image]

**Need to find out: Board revison that TheOldMan works on, and board revision that @sho3string works on**

@MJoergen @AnttiLukats:

* ["THE FIX" (see commit)](https://github.com/MJoergen/C64MEGA65/commit/ce927cb387150162837e71112ffe995fd870a561) is **only being applied in the binaries for the R6 board version** of the 5.1 release
* The binaries of the R3/R3A board versions have been created already at May 16 and do **not even** contain the "Sample RWDS a clock cycle later" HyperRAM fix, as shown in the following screenshot
* Since version 5.0 was rock solid regarding HDMI/HyperRAM on R3/R3A machines, we need to do a diff, in case either TheOldMan or @sho3string experiences this on R3/R3A machines

[image]

sy2002, 2024-06-30:
@amb5l FYI

sho3string, 2024-06-30:
[image]

After converting the Green Beret MiSTer core from M2M V1 to V2, the game exhibits jail bars after pressing space at the  
Repo is here: https://github.com/sho3string/RushnAttackMEGA65_r3_r6

The game otherwise runs on the M2M V1 framework and is playable albeit with some issues relating to DIP switch settings but completely unrelated to the above.

sy2002, 2024-07-01:
## Summary of the current situation:

a) We have a C64 V5.1 user on a R6 board "TheOldMan" who sees the "barcodes". Please note that the 5.1 release consists of 2 types of binaries. Binaries for R3/R3A, R4 and R5 do not contain your RWDS fix and do not contain Antti's fix.  Only the R6 binaries do contain the fix.

b) We did not hear from any R3/R3A, R4 and R5 user with any HDMI problems. We only heard from one R6 user who has problems. This is expected as HDMI/HyperRAM on R3/R3A, R4 and R5 always was rock solid - this is why I did not appliy the RWDS fix and Antti's fix before releasing the V5.1 core.

c) Muse/Sam/sho3string: It looks like the fixes for R6 also "destroyed" stability for R3. The V2.0.0 M2M contains Michael's RWDS fix and Antti's fix for all boards. We never tested these fixes on R3, since I only did build R6 when releasing the V5.1 C64 core and left the binaries for R3/R3A, R4 and R5 alone (i.e. without the  RWDS fix and without Antti's fix). Now Sam confirms that he sees the barcodes on R3 boards.

SUMMARY:

1. The RWDS fix and Antti's fix made the R6 board somewhat more stable but it did not fix the issue.
2. The RWDS fix and Antti's fix made R3/R3A boards more UNSTABLE as muse points out.
3. Luckily I did not apply the RWDS fix and Antti's fix to R3/R3a so that not everybody is affected in release V5.1 but "only" R6 board owners

sy2002, 2024-07-01:
@AnttiLukats: Do you have advise - for example, does it make sense to tweak the IDELAY constant? You changed this from 20 to 12 in your fix. Not sure whether we should tweak it up or down though, if at all:

https://github.com/MJoergen/C64MEGA65/commit/ce927cb387150162837e71112ffe995fd870a561#diff-be2f3034c39a2edc9f50cff2e2bdbddb942305f2f8016eb3d7add013748a8ad8L58

sy2002, 2024-07-01:
Interesting idea from @amb5l via Skype:

> OK, I have a suggestion. This will not be a small undertaking, but the result should be worthwhile. For an example of this approach, see the HDMI input of my “TMDS_CAP” HDMI analyser design.
>1)	Ensure that the controller can run at full speed (82MHz?) or half speed (41MHz?), and that this speed can be changed dynamically – that an individual transaction request includes a speed setting.
>2)	Write a “checkerboard” (alternating 1-0-1-0) pattern at half speed into HyperRAM.
>3)	Read the checkerboard pattern back repeatedly (at full speed), and train the read input IDELAY value on it. At the end of training, set the IDELAY value to the middle of the data/RWDS eye.
>This will give you a design that will adapt to PCB and device variation.

AnttiLukats, 2024-07-01:
Hi,

the "BUFR + IODELAY12 fix" made a design that I am testing on CR00107 Spartan7 very stable. Before the fix it was failing on one single type of RAM check: increment test passed, walking one passed, walking zero passed, constant 0xAAAA5555 passed and incrementing address failed. After changing IDELAY to 12 all tests passed, and are passing for 7+ days continued testing. When the error occurred it was always one single bit flip read out once wrong, next read to the same location returned already correct value.

If my fix makes R3 to show stripe pattern then this may already lead to the solution. The BUFR IDELAY12 fix I would say can already introduce "read wrong bit once" type error. The delay 12 should be pretty good, I do not think that IDLEAY tuning can fix the stripe error. The stripe error must be somewhere else, possible not at all in HyperRAM IP core. Some stupid timing issue that happens.

sy2002, 2024-07-01:
@AnttiLukats Thank you - the problem is that the fix is also not stable on all R6 boards. See above. We have at least two R6 users who have the original barcode problem. We thought we had solved it because none of us, including @AmokPhaze101 who did multi-day tests and including Deft could reproduce it any more. Scroll up, read "Summary of the current situation. And this "barcode" problem occurs because the ASCAL component that is in our HDMI rendering pipeline fails due to a locked up HyperRAM. So we are sure that this is a HyperRAM issue, but we are not sure why.

For your benefit - current thoughts from Michael: 
[image]

AnttiLukats, 2024-07-01:
you say locked up HyperRAM what you mean by this? That the RWDS is not detected and transaction on memory bus is not completed at all? This would yield very constant pattern, but probably on one transaction only, as the IP Core would recover on the next read. 

Stopped clock should not be a problem. Dynamic latency should also not be the problem. Not ideal IDELAY should not be the problem.

To me it seems that UNRELATED change in the code makes the bug to appear more or less. This is an indicator that we have somewhere an timing issue. Can you increase the main clock timing constraint say by 10% and retry? Does it make a difference?

AnttiLukats, 2024-07-01:
BARCODE ERROR:
============
* If there is something wrong with hyperRAM DATA latching, (IDELAY) then we would see random flicker on screen not barcode!
* some other bug with HyperRAM IP Core: we would not see a barcode!
*  HyperRAM memory internal problem (refresh?): we would not see a barcode!

Thing is that HypeRAM IP core can not return barcode data as data read from HyperRAM. The fact that we see barcode, means the bug is somewhere else.

MJoergen, 2024-07-01:
The barcode error happens because the HyperRAM IP Core goes into a lock up. What happens is that the Avalon Memory Map interface uses burst mode, and if the IP Core does not return the correct number of words as requested, then the interface locks up (the client connecting to the IP Core is waiting indefinitely).

So the question - as I see it - is what could cause the HyperRAM IP Core to not return the correct number of values? One plausible cause I can see is the sampling of the RWDS signal to determine the latency of the transaction.

AnttiLukats, 2024-07-01:
do not get this, does it lockup forever? Where does the barcode data come from then? The RWDS sampling for latency check is hard to get wrong, I do not think that it is a problem.

There is some time CSN must be deactivated between requests, is that OK with your core? But this also is not likely issue.

I must go, I will think about this some more.

MJoergen, 2024-07-01:
The barcode is generated by the HDMI rescaler, which uses the HyperRAM as a frame buffer. The rescaler continuously reads frame data from the HyperRAM into a small RAM, and generates the output display from this RAM. So if the data in the RAM does not get updated (due to the HyperRAM IP core lockup), then we get these barcodes.

Yes, the Avalon Memory Map interface protocol does not specify a timeout. So if the rescaler requests 64 words in a read request, and only gets 63 words, then the rescaler will just wait forever for the last word.

AnttiLukats, 2024-07-04:
The problem is in hyperram_rx.vhd file, at the very end (last line). The code as it is causes the statemachine in the controller module to freeze on the first time when the FIFO is empty. So the code as is only works if ALL data from the FIFO comes out as one chunk without any caps. One cap in FIFO stream, and STALL.

fix:

   -- This skips the first clock cycle of data from the FIFO.
   ctrl_dq_ie_d_proc : process (clk_i)
   begin
      if rising_edge(clk_i) then
         ctrl_dq_ie_d <= ctrl_dq_ie; -- delayed version of data valid
         ctrl_dq_ie_d2 <= ctrl_dq_ie_d; -- we need past-2 time also 
      end if;
   end process ctrl_dq_ie_d_proc;
   -- if it was low for 2 clock cycles then we cut out the valid signal
   -- it works as long as there is never 2 clocks without fifo valid
   -- but this is so by design, there is max 1 clock cycle where fifo data is not available
   ctrl_dq_ie_o <= (ctrl_dq_ie_d or ctrl_dq_ie_d2) and ctrl_dq_ie;

this is maybe not the best way, but the code still works and the bug should be gone.

sy2002, 2024-07-07:
@AnttiLukats  Thank you for continuing to support us. @MJoergen is currently indeed investigating the FiFo, but he is not 100% convinced yet, that your proposed solution will work. For your convenience and for tapping into your brain as well, I am copying a group conversation between @MJoergen, @amb5l and me that happened on Skype. Thoughts and feedback welcome:

[image]

**MJoergen:**

Status report:
I tried injecting errors into the RWDS sampling, and that provokes several different error conditions:
1. If during a read, the HyperRAM outputs a 1 on RWDS, but it erroneously gets sampled as 0, then the controller locks up. Because a long latency is mistaken for a short latency, and the HyperRAM controller is thus missing four clock cycles (= difference between long and short latency). So a burst transfer of 64 words only forwards 60 words.
2. However, if instead it's a write transaction, then data is lost. This is confirmed in the data sheet write data must appear at the exact correct clock cycle after transaction start.

Point 2 above means it is imperative that the controller correctly reads the RWDS. There is no possibility of work-arounds (e.g. by adding extra clock cycles) to mitigate an incorrect sampling.

Here is the relevant page of the datasheet:

[66-67WVH8M8DALL-BLL.pdf](https://github.com/user-attachments/files/16119007/66-67WVH8M8DALL-BLL.pdf)

[image]

This ILA trace shows a read transaction with long latency. It's captured using a 200 MHz clock, so that we can see the transitions on the RWDS signal. Some signals cannot be captured by the ILA, because the output register is placed in the I/O pad and cannot be connected to the ILA.

In this build, I've made use of the "falling edge + rising edge" two-stage synchronizer. So you see there is a latency of a 10 ns from "rwds_in_delay_idelay" to "hb_rwds_in_i".

The HyperRAM clock is generated by an ODDR fed from "hb_ck_ddr_o". There is a 10 ns latency from the "0 -> 2" transition of the "hb_ck_ddr_o" signal and the first rising edge of the HyperRAM clock. This is not seen in the ILA trace.

The yellow marker in the ILA trace is the sampling time of the "hb_rwds_in_i" signal.

When looking at the ILA trace it seems close / marginal. I'm comparing the sampling time at time 2048 with the falling edge at 2050, which is a difference of just 10 ns, i.e. 1 clock cycle.

In other words, it would seem "more" stable to delay the hb_rwds_in_i signal by one clock cycle, but that is exactly what we're already doing in the released version. As mentioned above, this trace is with the "faster" synchronizer.

I'm beginning to suspect that we (I) have not fully understood the problem (yet).

**amb5l**

OK I am thinking about this. You've seen Antti's suggested fix? He suggests the read data fifo is going empty and that stalls the Avalon burst which causes a crash.

**MJoergen**

I couldn't quite follow his reasoning. Nor do I see how the read data fifo could suddenly go empty. But as mentioned, I may be missing something.

**amb5l**

Does the Avalon side of the FIFO wait for a "not empty" flag or does it just start reading at a fixed point in time (perhaps determined by RWDS/latency)

**MJoergen**

It waits (indefinitely).

In the HyperRAM example design (separate repo) I've added some error signals (FIFO overrun/underrun, etc). But that's only helpful if we can reproduce on our own board.

**amb5l**

Of course. I hope (sort of) that my MEGA65 will exhibit this issue

So let me understand. Your controller causes clocks and address/command bytes to be fired into the HyperRAM for a read cycle, and - for the first 3 clocks (6 edges) the RWDS signal is driven by the memory to signal latency. You sample it - presumably as you get to the end of driving the address/command - to decide what to do next. Why, given that RWDS will have been driven for a while, do you pass it through a synchroniser?

**MJoergen**

Not any specific reason. More a case of trying to follow "best practices". 

But I get your point.

**amb5l**

You may need a timing exception in your constraints if you go asynchronous (I think you should try this). Otherwise the tools work to close timing on an assumed synchronous RWDS input. Perhaps FALSE PATH to the registers in the state machine that sample it.

**MJoergen**

But will it make any difference? My reasoning is as follows: In the released version there is a two-stage synchronizer. If I remove that  and simultaneously sample the signal two clock cycles earlier, the net result should be the same, right ?
[... 37 lines cut ...]
Here is a trace with a "long" latency.

I'm logging out for now, but feel free to write your thoughts, and I'll look into it tomorrow.

**amb5l**

1) Given where you sample RWDS, making the synchroniser delay shorter could actually be a problem. I think you need to leave it at 2 cycles. 2) If noise gets onto the src_clk_i, and that clocks the FIFO, then the FIFO could go mad. Corrupt pointers and flags etc. A clock enable should suppress this - is there one for ALL the flipflops that are clocked by src_clk_i?

I will keep thinking.

**MJoergen**

Ok, I have a new theory: I'm looking at the last waveform, in particular the Receive FIFO. It seems to me that there is a rather long delay from "src_gray_wr" to "dst_gray_rd". Specifically, I'm concerned about a FIFO overrun.

[... cut ...]
I'm trying to demonstrate this theory in simulation. Right now I'm surprised this works at all, i.e. on any boards. It seems to me (based on the simulation waveform) that it should fail all the time, at least for burst transactions.

Regarding possible solution, I'm considering two possibilities: Either reduce the synchronizer length on FIFO data and control signals (to one and two respectively), or increase the FIFO depth from two to four words.

**am5bl**

Interesting theory. I like the idea of a four word FIFO, personally.

AnttiLukats, 2024-07-08:
long story. But there really is a bug that locks up when FIFO is once EMPTY. and my fix fixes it, so why not test it out?

sy2002, 2024-07-08:
@AnttiLukats Yes, your proposed fix is independent from @MJoergen 's Odyssee. Also the four word FIFO idea from @MJoergen makes sense and might improve the situation. So what we will do is: Create a test build and work with the two guys on Forum64.de who can reproduce the problem to test it. This will take a few days. Please stand by for results. Thank you again for your persistence :-) and for your support.

AnttiLukats, 2024-07-09:
been thinking some more. FIFO depth 32 is for sure not required. I guess FIFO deep 2 is sufficient as well. FIFO depth 3 is 100% sufficient, FIFO depth 2? Possible also OK.

AnttiLukats, 2024-07-09:
good! I was thinking and drawing and doing excel and I think that in reality 2 deep FIFO would also work. There is theoretical possibility that FIFO occupancy would be 3 in the case that RWDS jitter during single burst is 5nS, this is not happening on real hardware so, I guess 2 level would be also ok. But if we have 4 level we are for sure on the safe side!

sy2002, 2024-07-09:
@AnttiLukats As you have seen above: Michael did an Alpha release that includes your "Empty FIFO" bugfix and expands the Receive FIFO to 4 words: https://github.com/MJoergen/C64MEGA65/commits/V5.2A1/

What is next: We will test the core internally and then share it with the folks on Forum64.de who can reproduce the barcode problem on their machines.

paich64, 2024-07-09:
Testing has just started :)

sy2002, 2024-07-10:
@paich64 Please share this file with TheOldMan and mgreima on Forum64 and either (a) invite/motivate them to document their results directly here in the GitHub issue (creating a GitHub account is easy) or (b) please do capture their feedback in the Forum and document their results here:

[C64MEGA65-V5.2A1-experimental-R6-boards-only.zip](https://github.com/user-attachments/files/16158118/C64MEGA65-V5.2A1-experimental-R6-boards-only.zip)

I guess you cannot upload ZIPs to Forum64 so for your convenience here is a short-link to this very file which directly starts the download: https://t.ly/ujcev

paich64, 2024-07-10:
> @paich64 Please share this file with TheOldMan and mgreima on Forum64 and either (a) invite/motivate them to document their results directly here in the GitHub issue (creating a GitHub account is [...]

I did so, no feedback yet, i will keep checking and report here. I have also asked TheOldMan and mgreima on Forum64 to report here if they can.

Homezecke, 2024-07-11:
In the 64 forum I am TheOldMan.

I have now tested all affected games for the “barcode” bug with the new core 5.2. None of the errors occurred! There were no problems even with different resolutions. There were no barcodes visible in any game!

sy2002, 2024-07-11:
@Homezecke Thank you VERY MUCH for reporting this bug and for testing the workaround. Please continue to use this Alpha 1 release of the version 5.2 of the C64 core which is supposed to be released in 2025 (or late in 2024). Please do not pro-actively encourage people to use it if they are not encountering the "barcode" bug. Currently we assume it is less than 1% of the R6 population who are affected by this glitch of our HyperRAM controller. We will need quite some additional time to find/fix the root cause and/or run through our release test protocol, therefore this Alpha release is only meant as a "painkiller" for people like you and not for the broad public.

I wish you a lot of fun with the C64 core :-)

paich64, 2024-07-13:
> @MJoergen and @AnttiLukats Fix works now for two users who experienced issues: @Homezecke from Forum64 and "Sidde" from Discord, see screenshot:
> 
> <img alt="grafik" width="1128" src="https://private-user-images.githubusercontent.com/6825267/347764363-c9d1c27b-a58d-44a5-af83-7744b9bc64b8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA4NjIyMTMsIm5iZiI6MTcyMDg2MTkxMywicGF0aCI6Ii82ODI1MjY3LzM0Nzc2NDM2My1jOWQxYzI3Yi1hNThkLTQ0YTUtYWY4My03NzQ0YjliYzY0YjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MTNUMDkxMTUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGE1MTQ0MGNmMWQ2YmQ4NDRjMjk0MGY0NmFjNGVlOThmMzFhZmI0ZDE0ODMzMmE1ZjY0MmQ3NWM5ZjE5N2FlMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.C1JFtSxGTp8sD9C6ShegGxsbqKtwJkt2YKP-aoXW5kg">

We just got confirmation that the issue is also fixed for @mgreima ! 

[image]

sy2002, 2024-07-13:
Awesome! Thank you @paich64 for supporting the Forum64.

I documented next steps on Discord:

[image]
see also this link: https://discord.com/channels/719326990221574164/801767398675316756/1261683009451462709

<br>And on Forum64:

https://www.forum64.de/index.php?thread/125676-c64-core-f%C3%BCr-den-mega65/&postID=2165804#post2165804

And with that I am closing this issue. @MJoergen and @AnttiLukats and @amb5l THANK YOU for your outstanding work.

sy2002, 2024-10-30:
Reopened to track the progress of adding this to the 5.2 maintenance release.

paich64, 2024-12-23:
@sy2002 @MJoergen I have been reminded that this issue also exists in Arcade core, and as far as i understood, the issue might be a more general issue (not only affecting the C64 core) which exists in the m2m framework. Are you going to fix the C64 core only or more generaly the m2m framework (with which, the C64 core and all other ported cores are made) ? Thanks.

sy2002, 2025-01-01:
@paich64 Happy New Year :-)  Indeed you are right, this is a more general issue that has been fixed by @MJoergen in the M2M framework. I need to do a release V2.0.1, soon and then everybody who uses the M2M framework for Arcade cores just needs to re-synthesize their cores (given that they are already von M2M V2.0.0).

@MJoergen Sorry for not being more responsive. Is https://github.com/sy2002/MiSTer2MEGA65/tree/V2.0.1 the right tag for an official M2M maintenance release V2.0.1?

paich64, 2025-01-01:
@sy2002 Thanks a lot ! happy new year to all of you !

By the way @sy2002 , did not forget you, tomorrow morning i will ship the "Super Snapshot V5 - snappyrom edition" which requires REU to be enabled so that we can use the REU version of Turbo Macro Pro which is embedded in this cartridge. You should get it by next week obviously, i will send you by email the tracking information ;)

MJoergen, 2025-01-03:
@sy2002 
> @MJoergen Sorry for not being more responsive. Is https://github.com/sy2002/MiSTer2MEGA65/tree/V2.0.1 the right tag for an official M2M maintenance release V2.0.1?

Yes, V2.0.1 is ready for release. It contains the fix for the HyperRAM lockup issue.

sy2002, 2025-02-24:
> [@sy2002](https://github.com/sy2002) [@MJoergen](https://github.com/MJoergen) I have been reminded that this issue also exists in Arcade core, and as far as i understood, the issue might be a more [...]

@paich64 I know, you have seen it in Discord. I am just also documenting it here for folks:

https://discord.com/channels/719326990221574164/801767398675316756/1342825651366789161

MiSTer2MEGA65 V2.0.1 which fixes this issue has been released:

https://github.com/sy2002/MiSTer2MEGA65/releases/tag/V2.0.1

We will keep this very issue open, because the C64 core which fixes it, has NOT YET been released. Version number will be 5.2 and for now C64 core users which encounter the problem still need to use the pre-release: https://t.ly/ujcev

MJoergen, 2025-03-16:
This should be fixed in V5.2A2

sy2002, 2025-03-16:
@MJoergen Wasn't this already fixed in V5.2A1?

MJoergen, 2025-03-17:
@sy2002 . Yes, you are correct.

sy2002, 2025-03-29:
This is known as fixed in V5.2, so I will close this one.

closed 2025-03-29 by sy2002