MJoergen/C64MEGA65 #146: PHI2 Clock Frequency

Kugelblitz360, 2024-07-12:
While trying to debug some cartridges (where a pure mechanical problem drove me nuts for a while), I have see that some freezers (Super Snapshot) do not work at all via EF3 or KFF, some (FC3) work better when the machine is NOT in Flicker Free mode.

I have used the Diagnostic tool from the KFF to display the Phi2 values measued there.

When the C64 core is in "Flicker Free Mode" (machine slowed down slightly to lock precisly on 50 Hz), Phi2 is measured at 982756. In this mode, most freezer cartridges will not come up on standard machine reset. FC3 will work after a Reset on the KFF, but has some flicker in the characters which goes away immediately when switching off Flicker Free Mode.

When the C64 core in NOT in "Flicker Free Mode" Phi2 is measured at 985218. FC3 works as it should. Super Snapshot still does not work.

When I keep the test running for a while, does values stay rock-stable.

Kim says on his website: "For reference the phi2 clock frequency should be around 985248 Hz for PAL."

I have asked Kim for suggestions on a Phi2 offset for KFF.

I am mainly wondering if my specific machine has a clock generator that is slightly off. What are the EXPECTED Phi2 values with the 5.1 core on R6? Can anybody measure their values (hold the menu button on the KFF for 2+ seconds) and compare with mine?

Thanks, Boris

Kugelblitz360, 2024-07-12:
AHHH! I see that these frequencies are identical to the ones shown as CPU speeds in the speed documentation, so my system must be right on target. Going to ask Kim for suggested offset values then.

paich64, 2024-07-12:
Hum, interesting, my frequencies reported in KKF tool are not the same as yours : 

Flicker-free off : 985186hz
Flicker-free on : 982724hz

Kugelblitz360, 2024-07-13:
Measurements made with different KFF firmwares - I was running 1.51, paich64 running 1.44

paich64, 2024-07-13:
> Measurements made with different KFF firmwares - I was running 1.51, paich64 running 1.44

I have tested (again) Super Snapshot V5 with my KKF using firmware 1.44

I could generate and save a snapshot to a .d64, basically all seems to work as expected.

I then updated my KKF firmware to 1.51 and here are the results : 
- the CPU frequencies are exactly the same as with firmware 1.44
- Super Snapshot V5 works exactly as it works with firmware 1.44 : No issue at all.

I have attached the Super Snapshot cartridge i have used. 

[Super_Snapshot_v5.22_v2_1990_PAL.zip](https://github.com/user-attachments/files/16201883/Super_Snapshot_v5.22_v2_1990_PAL.zip)

paich64, 2024-07-13:
@Kugelblitz360 for clarity purpose can you update the Issue title and mention the issue is observed on R6 ?

sy2002, 2024-07-13:
@MJoergen FYI and @Kugelblitz360 and @paich64 Thank you for your research and feedback.

Here is how I look at it:

* Multi-dimensional problem. For example: My R3 machine is able to run SuperSnapshot flawlessly on my EF3.

* Looks like a decision that users need to make: HDMI Flicker free versus using real hardware. Example is also IEC when they [attach real 1541s to the IEC port](https://github.com/MJoergen/C64MEGA65/blob/master/FAQ.md#switch-off-hdmi-flicker-free): Use real hardware, but then it is better to not use modern HDMI displays but stick with [analog CRTs](doc/retrotubes.md) and switch of HDMI-flicker free. OR, use emulated hardware and then use HDMI...

* Not sure if this problem is fixable at all, of if this is a "squaring the circle" issue. But maybe options to fine-tune the PHI2 phase shift might help.

* Also not sure if @MJoergen 's idea about a better HDMI flicker-free (see issue #119) might help

* In general this looks like a research task that then should lead to much better documentation on our side: **We could and should document (like a recipe book) what people could and should do when they want to do use case a, b or c.** 

I tagged this task as "research" and "documentation" and I am grateful for any additional research people make and document it here in the issue. Also if one volunteers to write the above-mentioned "recipe book" this would be related to issue #147 "Next Level Documentation"

Kugelblitz360, 2024-07-13:
Okay, so apparently there are tolerances between M65 boards putting out different PHI2 freqs. How can I contribute / run stuff on my machine to help debug?

Kugelblitz360, 2024-07-13:
One thing that I really want to try out now is different HDMI parameters besides Flicker Free (but can't get at my Mega until Monday morning). Are you running 720p 50 Hz or a different mode?

paich64, 2024-07-14:
> One thing that I really want to try out now is different HDMI parameters besides Flicker Free (but can't get at my Mega until Monday morning). Are you running 720p 50 Hz or a different mode?

I have tested 720p 50hz, 720p 60hz, 576p 50hz (both 5/4 & 4/3) - with & without flicker free - And it does not impact the frequency at all.

The only thing i observed is the following : 

2days ago these were my frequencies : 

Flicker-free off : 985186hz
Flicker-free on : 982724hz

Today these are my frequencies : 

Flicker-free off : 985184hz
Flicker-free on : 982722hz

15min later they increase a bit by 2hz and then become stable: 

Flicker-free off : 985186hz (+1hz)
Flicker-free on : 982724hz (+ 2hz)

At the same time, the temperature in the room is colder today. 

And we know from experience that all electronic components are sensible to temperature/humidity

I have no idea why we have this difference (you and me) when it comes to the core CPU frequency. And at the same time, I have no idea if such a low difference (~32hz) can make a real difference.

Kugelblitz360, 2024-07-14:
This actually is a totally different problem!

JaffyDOS + Core + KFF/EF3 does not work correctly. JaffyDOS (instead of JiffyDOS) has a different Reset routine.

I can't find any hints on the web that JaffyDOS would be incompatible with freezers, so this MIGHT be an issue that just appears on the Core, but not a real machine.

closed 2024-07-14 by Kugelblitz360