MJoergen/C64MEGA65 #148: Suggestion: PAL 50.125 HZ mode

Kugelblitz360, 2024-07-13:
Inspired by BMC64 (https://github.com/randyrossi/bmc64/blob/master/README.md) the ask is to have an additional HDMI mode that, instead of slowing down the machine, speeds up the HDMI output to 50.125 Hz. This is not compatible with all monitors but with some (as one of my cheap Medion monitors is proof of). The video logic of the RasbPi apparently can be set to such odd frequencies.

This solves the problem of some timing being off in flicker free mode but, again, would not be compatible with all monitors.

Cheers, Boris

sy2002, 2024-07-13:
Hi Boris, this one seems to be easy: Just switch off "HDMI-flicker-free" and all the HDMI outputs will be 50.125 Hz (even though the OSM claims they are 50 Hz, we might want to document this better). Closing this issue.

Kugelblitz360, 2024-07-13:
Absolutely - I thought you are actually dropping the frame in the M65 every six seconds, not changing the frame rate. As my monitor can do 50.125 I definitely should turn off flicker free then!

sy2002, 2024-07-13:
@Kugelblitz360 **Yes, you should indeed turn-off HDMI-Flicker-Free, if your Monitor can do 50.125 and everything should be buttery smooth. My favorite test is "Giana Sisters" Title Scroller: Let it scroll for 16 seconds. If you do not see jerky movements, then you're fine.**

And no, we are not dropping frames. What HDMI-flicker-free does is to [slow down the C64's overall system frequency to 99.75% of the real frequency](https://github.com/MJoergen/C64MEGA65/blob/master/doc/speed.md) and this leads to the VIC-II then outputting 50 Hz instead of 50.125Hz (and it leads to other effects, see PHI2 discussion).

Here are some more technical details just in case you're interested:
https://github.com/MJoergen/C64MEGA65/blob/master/doc/developer.md#visual-artifacts-and-dynamic-hdmi-flicker-fix

WebbedQuasar, 2026-02-14:
@sy2002 Yes, exactly. My monitor, which is an older LCD-TV, can display a non-standard 50.125Hz signal via HDMI from my MiSTer without any issues.

MJoergen, 2026-02-18:
This is related to #184

sy2002, 2026-05-05:
Continue this one (including the decision if we should do it) in #184 and closing this one.