MJoergen/C64MEGA65 #151: Mega65 R6, HDMI, Vertical bars

rharmonson, 2024-07-17:
With Mega65 R6 using either the v5.1 or v5.2a C65 core, I am getting vertical bars using HDMI on a Dell P1917S.

I played around with the differing monitor settings but they all exhibit the same symptom. I just have to wait. Sometimes it almost immediately. Other times, it takes a min or so.

paich64, 2024-07-17:
Hi @rharmonson 
can you please do this  
- delete all c64 cores installed on your Mega65
- install flash this core : https://github.com/user-attachments/files/16158118/C64MEGA65-V5.2A1-experimental-R6-boards-only.zip
- confirm you still get the issue or not ( as described in https://github.com/MJoergen/C64MEGA65/issues/145 )

rharmonson, 2024-07-18:
The file you linked has the same hash as the second core I tested and referenced v5.2a. I got it from:

https://github.com/MJoergen/C64MEGA65/issues/145#issuecomment-2219927361

SHA256: A204A23D2115BFB83D058FDE1356C3C09B550D688E3391335346D64212698353

I did have other cores installed. Specifically, Mega65, slot 0, and C64Mega65 v5.1, slot 1 ( and C64Mega65 v5.2A1, slot 2). I cannot imagine how that would impact anything but .. sure I will delete them, rewrite v5.2A and test. Give me a few minutes.

rharmonson, 2024-07-18:
Alright, I did the following.

1. Moved to a different and new sdcard; SanDisk Extreme 128 GB
2. Formatted sdcard using the Mega65; M + No Scroll
3. `curl -LO https://github.com/user-attachments/files/16158118/C64MEGA65-V5.2A1-experimental-R6-boards-only.zip`
4. `unzip C64MEGA65-V5.2A1-experimental-R6-boards-only.zip`
5. `mount /dev/sdb1 /mnt`
6. . `cp * /mnt && mkdir /mnt/c64 &&  mv /mnt/c64mega65 /mnt/c64/`

```
/mnt/
├── c64
│   └── c64mega65
├── C64MEGA65-V5.2A1-experimental-R6-boards-only.zip
├── C64MEGA65-V5.2A1-R6.bit
├── C64MEGA65-V5.2A1-R6.cor
├── hashes.txt
├── inofficial.md
├── README.md
└── VERSIONS.md

1 directory, 9 files
```

```
a204a23d2115bfb83d058fde1356c3c09b550d688e3391335346d64212698353  C64MEGA65-V5.2A1-experimental-R6-boards-only.zip
ac1da5b7586f1b212e5b9ac77ff36e828b4a7554943fd30ae62077b39a0cce6e  C64MEGA65-V5.2A1-R6.bit
8d273f031a9f3182d55eca5f9b5426d96c535345f6559728c2a383daaf87d687  C64MEGA65-V5.2A1-R6.cor
77f89e086200f56e06671b70af0a5bef9a340f18a617c61b54e140cd3f22d228  c64/c64mega65
```

7. `umount /mnt`
8. flashed c64mega65 into slot 1, power reset, and letting it bake for 15 minutes at the Commodore 64 Basic screen.

No issues at this point. In fact, it looks and is behaving great! Now to copy some games and test them.

Assuming all goes well, any idea why I had the issue? Seeing the zip file hash was correct, it was not a corrupted download.

rharmonson, 2024-07-18:
Closing this issue.

As much as I would love to understand the why, it seems to be working great now. Loaded several games and played without issue.

Thank you!

[image]

closed 2024-07-18 by rharmonson

sy2002, 2024-07-22:
@rharmonson Great to hear this is resolved. Here is my hypothesis: You had 5.1 installed in slot 1. Slot 1 is the "autostart" slot of the MEGA65 and maybe 5.1 was auto-started all the time and 5.2A1 was never started. When you deleted 5.1 and re-installed 5.2A1, then you made sure that we are starting 5.2A1. (Just an ockham's razor-ish thought.)

Completely independent from this issue here: Here is a warning about using a 128GB SD card:
https://github.com/MJoergen/C64MEGA65/blob/master/FAQ.md#4-sd-card-errors

> Is your card larger than 32GB? The core cannot handle SD cards larger than 32 GB.
If it works for you nevertheless, then go ahead :-)

rharmonson, 2024-07-23:
> @rharmonson Great to hear this is resolved. Here is my hypothesis: You had 5.1 installed in slot 1. Slot 1 is the "autostart" slot of the MEGA65 and maybe 5.1 was auto-started all the time and 5.2A1 [...]

A reasonably hypothesis. I was confident that I booted and selected the 5.2A1 rom. In fact, I did it several time, but, well, I cannot explain it so you are probably right. I just thought that I selected the 5.2A1.

I only have 64, 128, and 256 GB. They are SanDisk and Samsung brands. Thankfully, I have had no issues with the sd cards, yet, but 32 GB is more than sufficient capacity. I will look for a sale and buy a bunch of them for the Mega64.

Thank you!