MJoergen/C64MEGA65 #153: Feature Request: Please offer the ability to unload the currently loaded CRT image and return the CRT back to the "Load" status in the CRT selection menu

tonedef71, 2024-07-24:
Please offer the ability to unload the currently loaded CRT image and return the CRT filed back to the "Load" status in the CRT selection menu of the C64 Core.

Having such a capability would empower C64 Core users to clean up after themselves (with respect to loading/unloading CRT images anyway).

sy2002, 2024-07-25:
@tonedef71 Can you do a test for me using Version 5.1: Press <kbd>Space</kbd> when the OSD menu selector hovers over the CRT and tell me, if this already does the trick or not.

tonedef71, 2024-07-25:
@sy2002  For me, pressing the spacebar (while the `CRT:` menu option is highlighted in the OSD) has _the exact same effect_ as pressing the `RETURN` key (while the `CRT:` menu option is highlighted in the OSD): 
    --> the OSD pops up a file selection dialog to search the SD card for a `.crt` file.

sy2002, 2024-07-25:
Thanks @tonedef71 . We will look into this feature request for a future release.

sy2002, 2026-11-14:
Delivered with [WIP-V6-A22X3](https://github.com/user-attachments/files/34850669/C64MEGA65-WIP-V6-A22X3.zip). @tonedef71 feel free to rest and re-open the issue if it is not working as intended.

closed 2026-11-14 by sy2002