MJoergen/C64MEGA65 #157: Virtual Action Replay cartridge images do not work

WebbedQuasar, 2024-09-26:
I've been using a cart image of the latest Action Replay v7.5 for year now on my MiSTer. MiSTer's C64 can handle those images just fine: https://commodore.software/downloads/download/576-action-replay/11736-action-replay-v7-5-hacked
To launch a disk image, I mount AR 7.5, then attach the d64 file and press F7, F5 and F3. That's it.
On the Mega65, I get "?OVERFLOW ERROR IN 0" and crash when I press F7 to install fastload. Should be possible to solve this issue since it works on my MiSTer.

paich64, 2024-09-26:
Additional information : 

1) All .crt files available on https://rr.pokefinder.org/wiki/Action_Replay behave the exact same way on the C64 core : Trying to inject FAST LOAD results in "?OVERFLOW ERROR IN 0" error

2) The real physical cartridge (Action Replay V6) perfectly works with the C64 core : FAST LOAD injection works as expected and does not result in "?OVERFLOW ERROR IN 0" error

sy2002, 2024-09-27:
Thank you for reporting this @WebbedQuasar .
It is a known issue, we are tracking progress here in this issue: https://github.com/MJoergen/C64MEGA65/issues/69
So I am closing this one as DUPLICATE.

closed 2024-09-27 by sy2002