MJoergen/C64MEGA65 #161: TeensyROM not read correctly in C64 core.

Kugelblitz360, 2024-10-11:
When the MEGA65 is in M65 Core mode, GO64 activates the (physical) TeensyROM cartridge. It boots, Music plays, Menu works. Internal functionality like updating via SD card works too.

In the C64 Core the cartridge does not boot but the RAM test in Reset sees that there is a ROM and outputs `30719 BASIC BYTES FREE`

A quick and dirty test:
```
10 PRINT PEEK (32768)
20 GOTO 10
```
show the value 255 with a 70% chance. In the other 30% of cases it shows either 9 (the correct value for most carts), 63 or 127.
Doing the test with 32769, which most often contains $80, gets again most often 255, some 128, some 191 and some 159.

So somehow the cart simply is not read correctly.

sy2002, 2024-10-11:
@Kugelblitz360 are you OK to copy/paste your findings to this issue here: https://github.com/MJoergen/C64MEGA65/issues/143

And then we would close this one as duplicate and continue there?

One more question: Can you switch off HDMI Flicker-Free and test again?

Kugelblitz360, 2024-10-11:
Argh, I searched for Teensy and Github did not find it!!! Closed and moved.

closed 2024-10-11 by Kugelblitz360