MJoergen/C64MEGA65 #162: Action Replay IV physical cartridge doesn't work

CTalkobt, 2024-10-16:
I see in closed https://github.com/MJoergen/C64MEGA65/issues/157 that AR is working from an image, however inserting the physical cartridge into the port doesn't work. I'm met with the initial AR IV screen and garbage overwritten on the first 2 or 3 lines. 

Playing around with it, however - I have noticed that:
  - Holding down the reset button is necessary for it to boot into the menu properly. 
  -  Within the menu, I have have to continue to hold down the reset button to choose any of the options.
  - The only options that do work are F1) Clear Memory and F3) Reset (eg: back to basic). 

When Using F1 or F3 and getting into Basic, the  reset button doesn't have any function or action from that point forwards that I can see. 

Using the F5 (Utilities) or F7( Install fastload) fails to work. The behavior appears to vary based upon what I've been doing with the machine (eg: initial cold startup, having F1/cleared memory etc).

sy2002, 2024-10-17:
Thank you for reporting this issue. Are you using an R3 (pre 2024) or R6 (2024) model?

CTalkobt, 2024-10-17:
R6.

Last night however I did transfer the c64 roms over and created them in a slot.  With that slot it does work.  Expecting an out of box with non official roms I could see as unsupported.   I'll close it.

I may look at documentation to see if I missed anything initially or if it can be clarified a bit better with my initial understanding of the system.

Thanks,

closed 2024-10-17 by CTalkobt