MJoergen/C64MEGA65 #163: Easyflash Conversion of "Renegade III" does not work as virtual cartridge

Kugelblitz360, 2024-11-07:
[Renegade III The Final Chapter (by $olo1870) [EasyFlash].zip](https://github.com/user-attachments/files/17657058/Renegade.III.The.Final.Chapter.by.olo1870.EasyFlash.zip)

The attached EF3 version of Renegade III will work when running from a physical KFF or EF3 cartridge. When loaded as a virtual cartridge from the HELP menu, it will display loading and title screen and then stay at a black screen, never getting to the actual game.

Kugelblitz360, 2024-11-07:
[Renegade III The Final Chapter (by $olo1870) [EasyFlash].zip](https://github.com/user-attachments/files/17664456/Renegade.III.The.Final.Chapter.by.olo1870.EasyFlash.zip)
Newer version of that cart by the dev, still shows the same behavior: Works on EF3 and KFF but not when mounted virtually.

Kugelblitz360, 2024-11-10:
Additional info: When used as a Virtual Cartridge, the problem appears only when JiffyDOS is active. Switching to the standard Kernal makes the game work. Strangely it works with active JiffyDOS when it is a physical cartridge (either KFF or EF3).

sy2002, 2024-11-11:
@Kugelblitz360 and @MJoergen I propose to close this one due to the fact that it works without JiffyDOS - but you guys decide.

Kugelblitz360, 2024-11-11:
I need to test a bit more, have seen a second cart that only behaves badly as a SIMCRT but works fine as a physical; standard MagicDesk, so there is no reason why it should fail, but I need to do the full run of switching SD Cards, video modes, i.e.

sy2002, 2025-04-26:
Closed as NOT PLANNED because it works with the standard Kernal.

closed 2025-04-26 by sy2002