MJoergen/C64MEGA65 #165: Swiftlink (and compatible) Cartridge (physical) do not work on Core 5.1

Kugelblitz360, 2024-11-10:
Documentation suggests that we need bi-directional NMI, but I might be wrong.
[swiftLink-232-application-notes-revised.pdf](https://github.com/user-attachments/files/17692493/swiftLink-232-application-notes-revised.pdf)

Discord user xlar54 has raised this issue.

xlar54, 2024-11-10:
Just some additional context:

I connected my SX64 running CCGMS v6 and a Turbo232 cart to a C64 with CCGMS v6 and a swiftlink cart through a null modem cable.  They could talk to each other.

Put the same swiftlink cart into the M65 and booted up the 5.1 core.  Ensured i was using external cart in settings and loaded up CCGMS.  Systems could not communicate.

Happy to test if/when this is resolved.  Thanks!

sy2002, 2024-11-10:
@Kugelblitz360 and @xlar54 Thank you for opening this issue. Indeed, the core does not yet support bi-directional NMI. On the new 2024 MEGA65s (aka R6) bi-directional NMI is possible, on older revisions (pre 2024 aka R3/R3A) it is not. @xlar54 do you own a R6 or R3/R3A MEGA65?

I was indeed looking "for a reason" to do bi-directional NMI on R6 boards but never found a use-case why we need it / why it is worth the effort. Now we have an example use-case, so I am suggesting to plan it for our 2025 release called "5.3 Quality of Life" release.

oe4dns, 2024-11-10:
Just FYI - I have been using a Swiftlink compatible Link232-Wifi Cartridge and CCGMS to connect a Mega65 R6 to a Telnet BBS. For this I booted the current v5.1 C64 Core. (See [this Forum64 Thread](https://www.forum64.de/index.php?thread/151118-schwierigkeiten-bei-einer-mailbox-heutzutage-nicht-nur-beim-c64/&postID=2208481#post2208481) in German)

I am not sure if this applies here, but it definitely works good enough for Telnet connections.

Kugelblitz360, 2024-11-10:
Okay, will need to investigate whether there are hardware differences betweena real Swiftlink and the Link232 that would explain why one works and the other does not. There goes my bi-directional NMI theory.

Kugelblitz360, 2024-11-13:
Addendum from Travis/Sensorium: I know the original Swiftlink cart requires the dot_clock signal, but most other SL emulation carts don't.  Does M65 generate that signal (in synch w/ Phi2)?

sy2002, 2024-11-13:
@Kugelblitz360 Not sure what the M65 core does - but if you mean the C64 core: Yes we generate the dot clock signal in synch with phi2.

xlar54, 2024-11-19:
Thanks all.  I can confirm that an original CMD Turbo232 cart does indeed work under the C64 core.  My Swiftlink repro cart did not.   Unclear why, but perhaps that closes this issue?

Kugelblitz360, 2024-11-19:
The Mega65 outputs a bit more power on the cartridge port than a regular C64, that might confuse the electronics of that specific cartridge you have. I am closing this issue because it most likely is not a software/core issue, but we might want to retest that specific cartridge with a later revision of the Core.

closed 2024-11-19 by Kugelblitz360