MJoergen/C64MEGA65 #166: VERTICAL BAR OVER HDMI on R3 with core 5.1

paich64, 2024-11-24:
@sy2002 @MJoergen The issue is back again, but this time on R3 with core 5.1.

@JHerig is experiencing VERTICAL BARS over HDMI with his MEGA65 from batch 1 using the C64 core 5.1

Because of personal constraints he had to postpone his experience with the mega65 and now he is starting to use his machine and just discovered the issue.

The link of the original discord discusssion is here :
https://discord.com/channels/719326990221574164/719326990221574168/1310035903896354886

He has provided the following pictures: 

Mega65 model:
[image]

c64 core version:
[image]

The issue occures very quickly when he starts the core.

I'm waiting for him to confirm if this happens when he uses the standard kernal, since original pictures where specifically named "Jiffydos" so i need to be sure it's not related.

Unfortunately he does not have a VGA monitor so there's no way to confirm if vga is working find (presumably, if it's the issue we're well aware of, vga should be working).

We can't test V4 nor V5.0 because the latest stable Mega65 core loader fails to check CRC and won't flash these core versions.

If you see other tests that he could perform please let me know.

jherig, 2024-11-24:
To recap, this has happened with and without JiffyDos kernel, it happens approximately 60% of the time, and usually within 3 minutes of core boot up. The core appears to still be working, as sound effects etc still are happening and seem to follow what inputs you can supply and guess at what might be happening without visual reference.

paich64, 2024-11-24:
> To recap, this has happened with and without JiffyDos kernel, it happens approximately 60% of the time, and usually within 3 minutes of core boot up. The core appears to still be working, as sound [...]

@MJoergen @sy2002 exactly the same symptoms as the one we observed on R6.

sy2002, 2024-11-25:
@MJoergen Question: Is the fix that we did for R6 a "pure R6" fix, or could we (i.e. you) just build a bitstream V5.2A1 i.e. commit #f4df717bd6f7f7805f31ed361a271cc2227a8009  for @jherig to test?

MJoergen, 2024-11-25:
@jherig Please try this file:
[C64MEGA65_V5.2A1_R3.zip](https://github.com/user-attachments/files/17908300/C64MEGA65_V5.2A1_R3.zip)

jherig, 2024-11-25:
Installled and testing. First impressions are good (no repeats of the barcode).

sy2002, 2024-11-25:
@MJoergen Thank you for your support and for acting that fast. @jherig 's feedbeck sounds encouraging. Question: Is [C64MEGA65_V5.2A1_R3.zip](https://github.com/user-attachments/files/17908300/C64MEGA65_V5.2A1_R3.zip) "just an R3 build" of commit f4df717bd6f7f7805f31ed361a271cc2227a8009 or did you do anything else?   (Sorry for asking dump questions that I could theoretically research on my own, I am travelling a hell of a lot of very limited nerd time ;-) But I am still curious.)

MJoergen, 2024-11-26:
Sorry for not explaining. I did indeed just make an R3-build from the commit f4df717 .

sy2002, 2024-11-26:
@jherig Thank you for reporting this bug and for testing. Can you please test a week or so and report back here, if this fix leads to a stable and reliable core? Background info for you: @MJoergen and I are planning to release an official V5.2 "Maintenance" release end of this year / early next year that stabilizes a lot of aspects of the core. We **never** encountered the problem you have on any R3 board so far so this is a super important finding - also for making sure that V5.2 is the most stable release we ever had...

jherig, 2024-11-26:
After 1 day of testing (18 boots, approximately 3.5 hrs of runtime) there has been no repeat of the 'barcode' symptom. So things look very promising indeed as to this being a solution. Will keep testing.

sy2002, 2024-12-07:
@jherig I am curious: Is this fix still working for you and appears to be stable?

jherig, 2024-12-08:
After quite a bit of testing, the problem has not reoccurred. Things are looking quite good!

paich64, 2025-03-31:
@sy2002 i have been testing V5.2A2 since a dozen of hours now and the issue never re-appeared on both R3A/R6.

not directly related but vic20 core which was have regular jail bars when not generated from new framework also stopped having jailbars when generated with new framework (And i tested it dozens of hours).

So to me this issue is gone and is old story now.

closed 2025-04-07 by sy2002