MJoergen/C64MEGA65 #174: PRG loader should set $00BA to #08

Kugelblitz360, 2025-01-20:
The current PRG loader does not set $00BA so it is left at $00. It would contain the device number of the last device used. Some programs (like file browsers and some tools) rely on $00BA to be set to the device they were loaded from.

As there currently is no tape support, hard coding $00BA to $08 in the PRG loader should make some more programs work (but still will fail some copy protected tape games that expect $01 here).

I think the positive effects of getting some filebrowsers to work when called as PRG makes faking it as a #08 load useful although it is a "hack".

https://www.n0stalgia.org/common/pages/releases.php?op=showrelease&id=353

https://blog.worldofjani.com/?p=4172

paich64, 2025-01-20:
> The current PRG loader does not set $00BA so it is left at $00. It would contain the device number of the last device used. Some programs (like file browsers and some tools) rely on $00BA to be set [...]

I think it's already addressed in this issue : https://github.com/MJoergen/C64MEGA65/issues/144

Kugelblitz360, 2025-01-20:
No, this is different. Most "Injection" loaders that directly put PRG in to RAM do not set $00BA. This includes VICE (even 3.7) and TeensyROM (up to version 0.6.4) and Travis is going to add that to Teensy for one of the next releases and also the MIST code referenced does ignore it. And yeah, it is a hack and not real behaviour because we really never loaded from device #8. But some PRGs will crash because they expect it. Unfortunately this includes a lot of disk tools (and SD2IEC tools). Anything that will not access the last device, so basically any single load game (unless it checks if it was loaded from tape as a copy protection), will work without setting $00BA.

sy2002, 2025-02-22:
Thank you @paich64 and @Kugelblitz360 - I just had a phone call with @MJoergen and this will be part of V5.2 "Maintenance" and it is related to issue #144

MJoergen, 2025-03-16:
Should be fixed in V5.2A2

sy2002, 2025-03-29:
@paich64 Can you confirm that this one is fixed with V5.2A2?

paich64, 2025-03-31:
@sy2002 I don't know how to test this. @Kugelblitz360 can you please test the issue is fixed ?

Kugelblitz360, 2025-03-31:
This is fixed, all software I tested for this works great now.