MJoergen/C64MEGA65 #175: RESTORE key does not work as expected in SimCrt mode

sy2002, 2025-02-27:
[image]

sy2002, 2025-02-27:
[image]

sy2002, 2025-02-27:
The conversation on Discord begins here:

https://discord.com/channels/719326990221574164/794775503818588200/1343916500947304543

When scrolling down, one can see the discussion unfold.

sy2002, 2025-02-27:
It looks like an "obvious" bug, but we need to make sure, that cartridge_inst's output signal `crt_nmi` is not influenced by `restore_key_n` which we are feeding into `freeze_key_i`.

Side thought: I am not sure right now, if <kbd>Restore</kbd> works as intended as a "Freeze" key for simulated cartridges and I am not sure if it works as a "Freeze" key for real cartridges. Something we might want to investigate.

MJoergen, 2025-03-16:
Should be fixed in V5.2A2

sy2002, 2025-03-29:
@paich64 Can you confirm that this is fixed with V5.2A2, so that we can close this issue?

paich64, 2025-03-31:
@sy2002 yes it's fixed on both R3A & R6

closed 2025-04-07 by sy2002