MJoergen/C64MEGA65 #176: Insufficient Ultimax mode implementation for physical hardware — Incompatible with IDE64

mrjoso, 2025-03-24:
#### **Description**

In the current C64 Core on MEGA65 (V5.1), Ultimax mode in Expansion port is incomplete and prevents IDE64 cartridge from functioning properly due to memory mapping limitations.

---

#### **Problem details**

- It is **not possible** to map any external RAM/ROM to `$1000-$7FFF` in Ultimax mode.  
- This limitation prevents **IDE64** cartridges from working, as they rely on mapping external RAM from `$1000` to `$7FFF`.  
- **Incorrect behavior:** The `/ROMH` signal is generated for the `$A000-$BFFF` range, which is **incorrect** in Ultimax mode. However, this specific behavior does **not** interfere with IDE64 functionality.  
- **Correct behavior:** The `/ROMH` signal is generated for the `$E000-$FFFF` range, which is **correct** in Ultimax mode.  

---

#### **Expected behavior**

- The Expansion port should allow external RAM mapping to `$1000-$7FFF` in Ultimax mode, ensuring IDE64 compatibility.

sy2002, 2025-03-25:
Asked a question on discord, see screenshot:

[image]

sy2002, 2025-03-27:
[image]

sy2002, 2025-03-27:
@paich64 - this is FYI for you. The potential PLA change we talked about and that will NOT make it into the V5.2 release.

@mrjoso Let's approach this in steps. It has been quite a while, when I did my deep dive into the C64's PLA. Please have a look at this here:

https://github.com/MJoergen/C64MEGA65/blob/master/doc/PLA.md

and then have a look at this line of code, which SEEMS to be the culprit of the behavior you describe:

https://github.com/MJoergen/C64_MiSTerMEGA65/blob/5afb90e02fcefb053ac27792f42e742ad61e8b9b/rtl/fpga64_buslogic.vhd#L365C6-L365C24

You write that the /ROMH signal should never be generated for the $A000-$BFFF. And indeed, when I read the PLA.md (see above) one could interpret that you are right. So I created an experimental test core for you. It just comments-out the "culprit" line of code. It is for R3 machines-only and it is based on the soon-to-be released V5.2.

[c64mega65r3v52a2e1.zip](https://github.com/user-attachments/files/19494948/c64mega65r3v52a2e1.zip)

Can you confirm that the attached core V5.2E1 fixes what you called "Incorrect behavior" above, i.e. that it does not generate the /ROMH signal is generated for the $A000-$BFFF range any more?

If so, I will try to investigate the actual root cause for the problem, i.e. the inability to map any external RAM/ROM to $1000-$7FFF in Ultimax mode. Help from you is appreciated. Do you have documentation for me (weblinks) how this mapping is actually performed, or can you show me / explain to me the right PLA mechanisms for example using PLA.md from above and pointing me to it?

sy2002, 2025-03-29:
Feedback from @mrjoso on Discord:

I have tested the provided version 5.2.A2E1, and I can confirm that the /ROMH signal is no longer generated for the $A000-$BFFF range in Ultimax mode.

I’m not sure whether I should have paid attention to the following strange behavior in the provided experimental core. In the $A000-$BFFF area, internal RAM is mapped, whereas in Ultimax mode, there should be empty space (no RAM, no ROM if nothing is connected externally there). This behavior was not present in version 5.1.

That’s all for now. I have to stop here as I’m leaving for a week. I will be back to discuss the root cause. Thank you.

sy2002, 2025-03-29:
@mrjoso This core attempts to fix the problem and therefore should be able to support the IDE64 cartridge:

[c64mega65v52a2e2.zip](https://github.com/user-attachments/files/19521830/c64mega65v52a2e2.zip)

xlar54, 2025-03-31:
Tested for you.  The IDE64 runs fine on my Ultimate64.  But when I plug it into the M65 (using your core), it says "RAM" on the c64 screen.  I tried it a number of times, including removing and inserting.  according to the IDE64 docs, "RAM" at startup means:

Thecartridge RAM buffer does not work reliable. Clean contact
 edges, remove other cartidges or the whole port expander. Too
 long ShortBus cable might also cause this. Also you might try
 to reseat or replace the RAM chip.

mine is a 4.2, and i did clean the contacts.

Kugelblitz360, 2025-03-31:
http://baltissen.org/newhtm/x_dma.htm

has more info about the DMA cycles.

sy2002, 2025-04-01:
@xlar54 Thank you for testing - I highly appreciate your support.

@Kugelblitz360 Thank you for the clarity - super valuable insight. Then this whole thing is not the kind of quick PLA bugfix I originally intended to provide. It is more of a core enhancement project.

@mrjoso Thank you for reporting this. I was under the impression that by making sure we can map $1000-$7FFF and $A000-$CFFF in Ultimax mode, IDE64 would work. The experimental core V5.2 Alpha 2 Experimental 2 (V5.2A2E2) [c64mega65v52a2e2.zip](https://github.com/user-attachments/files/19521830/c64mega65v52a2e2.zip) is supposed to do exactly this. **It would be great, if you could confirm, that at least the memory mapping works as expected**, then this would be at least one step forward - and helpful for the future, so that I would know that the code in branch dev-fix-PLA is kind of correct.

Supporting DMA via Expansion Port is a bit of a larger project and not possible in my current time budget. But it is great that we now have a cartridge (IDE64) with which we can test it as soon as it is implemented.

mrjoso, 2025-04-02:
Hi, I need to correct some misconceptions.
IDE64 does NOT use DMA.
I will test the provided core when I get back. 
Message RAM after boot means that external RAM is still not available in Ultimax mode.
Joso

Sent by from my Android device.

1. dubna 2025 7:35, sy2002 psal:

@xlar54 Thank you for testing - I highly appreciate your support.

@Kugelblitz360 Thank you for the clarity - super valuable insight. Then this whole thing is not the kind of quick PLA bugfix I originally intended to provide. It is more of a core enhancement project.

@mrjoso Thank you for reporting this. I was under the impression that by making sure we can map $1000-$7FFF and $A000-$CFFF in Ultimax mode, IDE64 would work. The experimental core V5.2 Alpha 2 Experimental 2 (V5.2A2E2) c64mega65v52a2e2.zip is supposed to do exactly this. It would be great, if you could confirm, that at least the memory mapping works as expected, then this would be at least one step forward - and helpful for the future, so that I would know that the code in branch dev-fix-PLA is kind of correct.

Supporting DMA via Expansion Port is a bit of a larger project and not possible in my current time budget. But it is great that we now have a cartridge (IDE64) with which we can test it as soon as it is implemented.

—

@xlar54 Thank you for testing - I highly appreciate your support.

@Kugelblitz360 Thank you for the clarity - super valuable insight. Then this whole thing is not the kind of quick PLA bugfix I originally intended to provide. It is more of a core enhancement project.

@mrjoso Thank you for reporting this. I was under the impression that by making sure we can map $1000-$7FFF and $A000-$CFFF in Ultimax mode, IDE64 would work. The experimental core V5.2 Alpha 2 Experimental 2 (V5.2A2E2) c64mega65v52a2e2.zip is supposed to do exactly this. It would be great, if you could confirm, that at least the memory mapping works as expected, then this would be at least one step forward - and helpful for the future, so that I would know that the code in branch dev-fix-PLA is kind of correct.

Supporting DMA via Expansion Port is a bit of a larger project and not possible in my current time budget. But it is great that we now have a cartridge (IDE64) with which we can test it as soon as it is implemented.

Kugelblitz360, 2025-04-03:
Okay, I stand corrected. I was thinking of reading the IDE64 source code tonight, but there is a pretty good explanation on what is happening here: https://c64os.com/post/supportingide64
Looking into that I would guess (and will confirm with source code later) that the IDE64 reads stuff into its own RAM that is mapped partially into $8000 to $BFFF and then transfers stuff with fast unrolled loops from there into other memory areas. The code for that most certainly lies in $033C space because IDE64 can not load anything below $0400 and it makes sense that you have a copy loop that switches the RAM in and out in that memory area. My corrected guess for the C64 core now is that the RAM behaviour is not simulated correctly.

Kugelblitz360, 2025-04-03:
Okay, I tried to read the IDE64 sources. Let's say they are special and I understand why nobody really picked up this code for other projects. As the self test for the IDE64 fails, I have checked the "extramtest" routine in "selftest.asm" first. It basically copies some code into the low part of the stack that in fast loops checks whether RAM can be read at certain locations. Apparently the PLA has been told to access RAM in the cartridge, not in the C64 memory, and the test pattern that it is matched against would be in ROM at that point in time. As this test seems to fail (that's when "RAM" appears like mentioned by <@360851773130407938> ), writing to that RAM fails. There are no register or IO manipulations during the loop, so basically this is a fixed memory config used during the whole test cycle.

Trying to read the actual LOAD routine is a mess because there are conditionals and branches everywhere for Jiffy, SCPU and so forth, but the CF load seems to read a certain byte (data_g),  than copy this and another bytes into real C64 RAM, then trigger the data_g read again and get two bytes, so one loop in the stack gets you 4 bytes and the full routine up to 512 bytes (because it exits with a BPL so it does up to 128 loops) . Reading the data_g byte/register must somehow trigger the transfer of the next two bytes because there is no indirect adressing and also no label that would suggest that this code gets modified. It clearly reads data_g two times and gets two different values. This 512 byte transfer really does not need a lot of cycles, but it depends on that the IDE64 sees the read to the one address and then automagically provides the next while the CPU is writing the values. **Can anybody confirm this?**

`;-----------------------------------
; CFS load 512 bytes normal
;
load_lot_cfs_r	.proc
		.logical $101
                .if SUPERCPU=0
		sta cfg_stnd		;3
                .else
                bit $2c2c
                .fi
lp		lda data_g		;6

		#lda_low_reg		;9
at1		sta $0000,y		;12

		lda hig_reg		;15
at2		sta $0000,y		;18

		lda data_g		;21

		#lda_low_reg		;24
at3		sta $0000,y		;27

		lda hig_reg		;30
at4		sta $0000,y		;33

		iny			;34

		bpl lp			;36
		#sta_cfg_open		;39
		rts			;40
		.here
		.pend
load_lot_cfs_re`

mrjoso, 2025-04-05:
Title changed to remove incorrect references to DMA.

mrjoso, 2025-04-07:
Tested version 5.2.A2E2.
In Ultimax mode, data on the $1000–$7FFF is still floating, even with IDE64 external RAM connected. The cartridge asserts external /RAM_CS so the address lines + PHI2 output on the Expansion port are active.

sy2002, 2025-04-07:
@mrjoso And here are some additional thoughts that came up when I asked chatGPT for myself :-) Besides a lot of ["irrelevant stuff" that it output](https://chatgpt.com/share/67f37da8-5700-8001-b611-4f67d1f759a4), here are some relevant thoughts:

* "It’s important to verify that when IDE64 is active the external device really drives the data bus." <== I am not verifying this at all and always switch to "read data from cart" WHEN we are in the unmapped ultimax mode and the CPU is in read mode (see Step 3 above). Maybe I need to wait until IDE64 signals that it has data for the CPU? How does it signal?

* "Verify that this “modified” address aligns with what the IDE64 expects for external RAM mapping. If the IDE64 is looking for a direct one-to-one mapping in the $1000–$7FFF range, any transformation may cause a mismatch. It might be useful to simulate or capture these address lines (using a logic analyzer via JTAG, as suggested) to see if the output meets the IDE64 specification."

mrjoso, 2025-04-07:
Hi @sy2002, I’ve just ordered a JTAG adapter and I’m willing to install Vivado, but it will take some time.
I also need to get familiar with the source code first.

The “external /RAM_CS” signal is generated inside the IDE64 cartridge by combinational logic from A12–A15, PHI2, BA, _EXROM, and _GAME. It enables the external RAM and the data bus driver when the C64 CPU accesses any address in the $1000–$7FFF range in Ultimax mode.
It behaves similarly to the _ROML or _ROMH signals generated by the C64’s PLA, but for a different address range.
I was able to measure that the signal is asserted when running a simple program that periodically reads from that address range.

Thank you for the detailed explanation of what you did to support the $1000–$7FFF mapping, I’ll take some time to understand it and get back to you.

sy2002, 2025-04-09:
@mrjoso Thank you for your support! Happy to do a Zoom call or something like this, to stare at the code together and to brainstorm together, if you want to.

Hint/question: Is the BA signal relevant for the IDE64? The C64 core currently has an extremely high degree of compatibility with hardware cartridges even though (or because?) we are setting BA to be always `HIGH` (`1`) and not touching it at all. See this obviously kind of "discouraging" comment that I wrote a while ago:

https://github.com/MJoergen/C64MEGA65/blob/dev-fix-PLA/CORE/vhdl/main.vhd#L860

mrjoso, 2025-04-09:
@sy2002 The BA signal is relevant on the C64 for ensuring register or RAM write stability when VIC sprites come onto the scene. We're not at that stage yet.

However, I have an interesting observation from 5.2.A2E2 yesterday.

In Ultimax mode, the address range $A000–$BFFF still shows promising behavior.
I'm able to map RAM/ROM to this area — in contrast to $1000–$7FFF and $C000–$CFFF, where data floats when unmapped (and the CPU doesn't see external RAM/ROM when mapped).
In the case of $A000–$BFFF, data is pulled up when unmapped, which differs from the C64, where data floats in this area — but that’s just a cosmetic difference compared to the correct behavior with external RAM/ROM.

We can probably start from here.

mrjoso, 2025-04-09:
I tried to follow the $A000–$BFFF address bus path.
I'm not sure if the fpga64_buslogic.vhd file I'm looking at corresponds to version 5.2.A2E2.

Line 380 (Address Logic):
In Ultimax mode, when the CPU address bus falls within the range $A000–$BFFF, the signal cs_ramLoc is asserted '1'.
This occurs in the else branch.

Line 258 (Data Multiplexing):
If cs_ramLoc = '1', then ramData is assigned to dataToCpu.
This looks like that the CPU reads data from RAM when accessing the $A000–$BFFF range in Ultimax mode.
This behavior was present in version 5.2.A2E1.

sy2002, 2025-04-21:
@mrjoso thank you for starting to looking into this. If you want to check, if the version of `fpga64_buslogic.vhd` you looked at is correct, then use the branch `dev-fix-pla` because this is where everything related to experiments around this issue is located. And branch `dev-fix-pla` points to `C64_MiSTerMEGA65 @ 64c9314` so here is the correct file:

https://github.com/MJoergen/C64_MiSTerMEGA65/blob/64c93140801030a752b2ed66dbadc83f13776486/rtl/fpga64_buslogic.vhd#L370

It also contains the change I made to avoid that the ` /ROMH` signal is no longer asserted (see above).

I didn't (and won't in the next weeeks) have the time to validate your above-mentioned analysis, but assuming that it is correct, this would not be a problem, but exactly the behavior I wanted to achieve, because the fact that the C64 should read from the cartridge and not from RAM is not implemented in the MiSTer core (as this core does not support external hardware cartridges at all), so it is not implemented in `fpga64_buslogic.vhd` . Instead, it is implemented in our wrapper of the core, in `main.vhd`, see here:

https://github.com/MJoergen/C64MEGA65/blob/dev-fix-PLA/CORE/vhdl/main.vhd#L595

So the RAM access you are observing in `fpga64_buslogic.vhd` is overridden by an access of the hardware cartridge: 

```vhdl
      -- Access the hardware cartridge
      elsif c64_exp_port_mode_i = C_EXP_PORT_HARDWARE and (cart_roml_n = '0' or cart_romh_n = '0' or cart_ultimax_unmapped = '1') then
         c64_ram_data <= data_from_cart;
```

BUT: Since it is not working, there is a bug somewhere that we need to find together :-)

sy2002, 2025-07-05:
@mrjoso So I found the reason why we had the regression: Since I refactored the wrapper and also the signalling of the Ultimax unmapped mode in 5.2A2E3 compared to 5.2A2E2, I needed to explicitly set the unmapped flag in the $A000-$BFFF case. Also - inspired by your comment and your findings - I made sure that in this very Ultimax unmapped mode, the core reads "RAM". Important: It does not read "RAM" in reality, because the wrapper in main.vhd, which is deciding what "RAM" actually means then feeds data from the cartridge into the core in the Ultimax unmapped case. Have a look here:

https://github.com/MJoergen/C64_MiSTerMEGA65/commit/06719b7d3098d3f046f437484e2ce5437ef6dcb1

One additional question: In fact this code here treats not only $1000–$7FFF and $A000-$BFFF as unmapped but also $C000-$CFFF. I assume this is OK?

Thank you for your support and please re-test this attached version 5.2A2E4 of the core: 

[c64mega65v52a2e4.zip](https://github.com/user-attachments/files/21074897/c64mega65v52a2e4.zip)

mrjoso, 2025-07-05:
Sorry, I should have mentioned that the line numbers were meant as a rough guide.
I was pointing out the `else` statement that applies in both Ultimax and non-Ultimax modes when a RAM bank is active.

sy2002, 2025-07-05:
@mrjoso no worries. Looking forward to your tests of the new test release.

mrjoso, 2025-07-05:
@sy2002 Huge thanks for your effort. I will test it soon, but maybe someone else will get access to the hardware before me. :-) Looking at the code, it looks very promising. (I'm convinced that treating $C000–$CFFF as unmapped in Ultimax mode is correct - I didn't mention it earlier since it's not used by IDE64.)

Kugelblitz360, 2025-07-07:
Might this fix EF3 Kernal replacement?

sy2002, 2025-07-08:
@Kugelblitz360 No. To support EF3 Kernal replacements, we need to be able to sense the cartridge's modifications of address line A14.

There is a separate enhancement request for this, see #126

sy2002, 2026-02-15:
Done (merged to V6 develop).

closed 2026-02-15 by sy2002