MJoergen/C64MEGA65 #177: Chat64 cartridge issue

huijaa, 2025-04-28:
Cartridge Chat64 (from https://www.chat64.nl) don't work with a newest core C64MEGA65 V5.2 but with previous version V5.2A1 it's working fine. Can you look in Discord MEGA65? Thanks!

sy2002, 2025-04-29:
@huijaa Thank you for reporting this. Can you post the discord link to the conversation here? I don't know where to look in Discord, there are a lot of channels. Just paste the link to the beginning of the conversation here.

@paich64 Can you investigate and document here?

huijaa, 2025-04-29:
> [@huijaa](https://github.com/huijaa) Thank you for reporting this:
> 
> * Are we talking about a hardware cartridge here or a `.crt` ?
> * Can you post the discord link to the conversation here? I don't know where to look in Discord, there are a lot of channels. Just paste the link to the beginning of the conversation here.
> 
> [@paich64](https://github.com/paich64) Can you investigate and document here?

https://discord.com/channels/719326990221574164/1340705384121372792

It is a hardware cartridge with a wifimodem in. See chat64.nl

Kugelblitz360, 2025-04-29:
Looking at the ESP32 source code, might first suspicion is this: This cart intialised itself when the C64 is powered on and when it is ready it issues a RESET. Unfortunately on the MEGA65 this RESET simply might come too early? The Core needs to be loaded and started and if this takes longer than the Chat64 intialisation, the RESET will not actually reset the C64 core yet.
The timing might be so on the edge that this one version of the Core just made it.
The other source might be unusual NMI handling, I see that the cart does that a lot, and NMI handling did change because the Restore-Key did not work as intended, and that was fixed for 5.2, but then again, we'd need to understand how Chat64 (ab)uses the NMI line.

sy2002, 2025-04-30:
@Kugelblitz360 Indeed a very interesting thought: Might it be, that the cartridge selection mechanism is not "getting it" and therefore not initializing the C64 core at all. Let's test that:

@huijaa Some questions:

1. Can you describe what "doesn't" work means? Is the screen black? Are you seeing the C64 Basic Screen? 

2. When you press the "Help" key: Is the C64 core's showing up or are you in the MEGA65's C64 mode?

3. And just to rule out a common error: Is the C64 core configured to use HARDWARE cartridges? The core saves this behavior, i.e. if you ever switched to `.crt` loading, you will need to switch back to HARDWARE mode, before it will work with your hardware cartridge.

@Kugelblitz360 FYI: The Restore key topic is irrelevant for this issue now that we learned that we are talking of a HARDWARE cartridge. The Restore key topic is a pure `.crt` cartridge simulation topic.

paich64, 2025-04-30:
I will receive my own cartridge within a few days. So i will be able to perform the checks and figure out what's really happening from an end use perspective.

paich64, 2025-04-30:
@sy2002 here is a picture of what's happening when the C64 core is loading with the cartridge inserted.
It does not seem related to the boot process itself but to the cartridge starting : 
https://youtu.be/sxBtHHxWOUM

sy2002, 2025-05-01:
@paich64 Interesting video. But why does he need to select the C64 core manually. This already shows, that there is a cartridge selection problem at the very beginning, i.e. the cartridge is not yet identified as a C64 cartridge by the MEGA65's cartridge selector, right?

paich64, 2025-05-01:
@sy2002 they confirmed the cartridge is detected as a C64 cartridge : When they start the mega65 with the cartridge inserted, the C64 core starts and the cartridge loading starts but stops as in the video, this with v5.2 while obviously with v5.2A1 the cartridge perfectly starts until the end of it's startup sequence.

paich64, 2025-05-10:
@sy2002 I have received my CHAT64 cartridge. I tested it with my MEGA65 R6. It works neither with the CORE C64 V5.2A1, nor with the CORE C64 V5.2. In both cases, the CORE C64 starts up correctly, which can be verified by displaying the OSM menu. So the recognition phase of the C64-type cartridge works well, and so does starting up the C64 core, but it's after that that things get complicated: With Core V5.2A1, I go straight to the Basic prompt. With Core V5.2, like you, I have a phase during which the edges of the screen flash. Eventually, however, the flashing stops and nothing happens. Curious thing: when the flashing freezes, if I lower/raise the cartridge slightly, I leave the screen with the flashing blocked and end up with the BASIC prompt. I'm going to try it with a real C64, but as it is, whatever the core version, the cartridge doesn't work on my R6.